ControlSystemStudio/phoebus #1702: Question: Tracker behavior with multiple selection

georgweiss, 2020-12-17:
Currently the tracker box size is defined by the min and max coordinates of selected widgets. Any non-selected widgets within that box  cannot be selected in the drawing area, see screen shot.

Users are questioning this behavior with reference to other applications where multiple selection will render separate tracker boxes around the selected items only.

I'm not saying Phoebus does it wrong, but wonder about the rationale for the current implementation.

[image]

kasemir, 2020-12-17:
> Any non-selected widgets within that box cannot be selected in the drawing area, see screen shot.

You can. Command (Mac) respectively Ctrl-click (every other OS) within the tracker will add unselected widgets or remove already selected widgets.

The tracker uses the overall outline because 'resize' will affect that outline, and 'move' will move that outline. Besides, the most common operation is likely 'move', and this way you can move by dragging any part of the tracker, no need to hit within one of the potentially small selected items.

kasemir, 2020-12-17:
Using Ctrl- resp. Command-click to add/remove from a selection seems pretty prevalent.
BOY did that, PowerPoint uses that, so most people are already familiar with the idea.

georgweiss, 2020-12-17:
You're right, I did not know CMD/CTRL is the trick, neither did my user. My bad.

In any case, as you mention PowerPoint... Even though Phoebus mimics the behavior in terms of selection, move and resize, it is not obvious to what end the tracker is rendered the way it is. In particular as widgets included in the selection are marked with a reddish border.

kasemir, 2020-12-17:
Selected widgets are marked with a red border to, well, indicate what's selected within the tracker outline, and support adding/removing selected widgets via cmd/ctrl click.

The tracker uses the overall outline to be easy to drag and resize. The 'handles' on the corners and centerlines are comparably large for the same reason. When instead placing one tracker on each selected widgets, these handles are necessarily smaller and then hard to click. You want to resize a widget, but miss the handler, instead click on the background and thus lose the selection.

georgweiss, 2020-12-17:
OK, I see your point.
Thanks.

kasemir, 2020-12-17:
The idea of basing the tracker on the outline may be unfamiliar to PowerPoint users, but it's actually quite common in CAD packages where fine-tuning the selection becomes important:

[image]

georgweiss, 2020-12-17:
I agree comparisons need to account for additional applications besides PowerPoint. However, whether we like it or not, the Microsoft tools tend to set the standard for various aspects of software behavior.

georgweiss, 2020-12-21:
Reopening as user reports issue.
In the attached screen shot three labels are placed on top of a background image. If the labels are selected in the indicated order, the last one cannot be selected as the background image is selected instead.
[image]

kasemir, 2021-01-04:
In this example it seems pretty straight forward to select the top, then the middle and finally the bottom label, which works fine.
Finally, there's the outline/tree view.

georgweiss, 2021-01-04:
Yes, selecting in that order works. But top and bottom are selected first, middle cannot be selected.

georgweiss, 2021-01-12:
@kasemir , can you please comment?

kasemir, 2021-01-12:
All I can say is that deciding what to select in a 'click' when several widget overlap is hard.
I've described two ways to accomplish what you want:
Click in top, middle, down order, which you confirmed works.
Or use the tree view as an aid.