ControlSystemStudio/phoebus #1711: Preference view

shroffk, 2020-12-21:
With multiple beamline and multiple sets of preferences becoming quite common at NSLS2 a simple UI would be quite useful.
The current goal is to work on a read only view.

shroffk, 2020-12-21:
[image]

krisztianloki, 2020-12-22:
A future feature could be to make the individual preferences available as PVs so a "Preferences UI" could be created with Display Builder itself

kasemir, 2021-01-04:
Note that you can also view the preferences, in a format that you can copy/paste into your own settings.ini file, via Help, About, Show Details, Preference Settings

georgweiss, 2021-01-18:
Any plans to merge this?
Asking as we are rolling out distribution of Phoebus from IT management tools. Users would then need a convenient way to edit preference settings as command line will not point to an ini file.

shroffk, 2021-02-10:
This is merged. You will have to include this in the site specific product

closed 2021-02-10 by shroffk

georgweiss, 2021-02-11:
OK.
Any plans to support simple edit (i.e. restart required)?