ControlSystemStudio/phoebus #1715: Question: reload behavior of local PV

georgweiss, 2020-12-23:
When an OPI is reloaded at runtime a local PV will be "reset" to its initial state/value. However, if the same local PV is used by another OPI, its value is set to the current value of the PV.

My question is if this is intentional and I'm asking at it might be a bit confusing when after reload the value of a PV depends on other OPIs.

shroffk, 2020-12-23:
This is by design.
All pv's with the same name represent a connection to a common signal. Thus they all have the same value, update at the same time, have the same timestamp, etc.
This is true for channel access pvs and the behaviour is same for local pv's too.

To have local pv's which are specific to a single opi users should use the ${DID} macro in the name
loc://test_${DID} 
This will ensure that loc pv's will be unique

georgweiss, 2020-12-23:
Thanks, but this is not so much about uniqueness as about how a local PV behaves when OPI is reloaded. In case the local PV is shared by multiple OPIs, reloading one OPI will not change the value in that OPI. But if the local PV is not referenced in any other OPI, the value is "reset" by a reload operation.

shroffk, 2020-12-23:
Local pv's are cached and the same pv is used in multiple opi's.
As long as there is at least one reference to that local pv, the pv and its last value is retained.

When there are no other opi's referencing that local pv except for the current screen...a reload will result in the last reference to the local pv to first be removed, the local pv is then deleted, then the opi is run again during which time the local pv is now recreated and initialized to the default value.

georgweiss, 2020-12-23:
And this is what I indeed observe. However, an operator might not know the underlying mechanisms, and might not know that the value presented by a widget is a local PV that is set to a default value when the OPI is reloaded. Since the context menu offers the possibility to reload the display we should expect it to happen occasionally, whether it makes sense or not. The user might be confused when a widget displays a different value than it did before the reload operation. 

But maybe this is more a question on OPI state and design in general, so this could well be a matter of clarifying the potential side effects of a reload operation.

shroffk, 2021-01-04:
How should this ticket me closed? Should we add a tag for future reference since this behaviour is not going to change

georgweiss, 2021-01-04:
We can close it. 
I will check if we need to clarify this a bit in the documentation.

closed 2021-01-04 by georgweiss