ControlSystemStudio/phoebus #1719: Windows network drive file path for button

tynanford, 2021-01-06:
I am having some issues working on getting an Action Button widget to open a new display, when the display path is on a networked drive on a Windows machine.

I have been able to get the Phoebus home display button and top resources menu to open the networked files, but have issues with File->Open and with the button.

As an example:

The file: //my-server/home/user/file.bob

org.phoebus.ui/home_display=file:////my-server/home/user/file.bob

For the action button Display Path, I have tried the following:
file:////my-server/home/user/file.bob
////my-server/home/user/file.bob
//my-server/home/user/file.bob
\\my-server\home\user\file.bob

I also notice that right clicking a running display on a network path and selecting "Display Editor" reports an error.

Thanks

kasemir, 2021-01-06:
Well, naturally requires developing/running in Windows to reproduce, while I use Mac and Linux...

> org.phoebus.ui/home_display=file:////my-server/home/user/file.bob

So it does open that 'home' display, the problem starts when you then try to open a related display?
What is the exact error when you use a relative path, like "x.bob" for file:////my-server/home/user/x.bob in the same folder, or "../../share/sub/y.bob" for file:////my-server/share/other/sub/y.bob ?

tynanford, 2021-01-06:
My life would be much better if we didn't have to support the windows users :)

I added a couple of screenshots. Using relative paths did not seem to fix the issue for now.

[image]
[image]

The 'home' display only works when I open it from the Toolbar "Home" button. In addition, if I exit Phoebus with this 'home' display open, and re-open Phoebus, I get errors that it can't find the file.

I bring this up because maybe the Home button in the toolbar uses different code?

tynanford, 2021-01-06:
I also was able to set up the File Browser to point to the networked drive. Users are able to open files from the file browser in display_runtime, but display_editor gives the error above: Cannot load model from C:\file\path

kasemir, 2021-01-06:
Meanwhile, for what it's worth, over here we access common files via a web server.
Displays are accessible as http://server/displays/some/path/file.bob, and that's the same for Windows, Linux, Mac, instead of having to deal with Samba vs. NFS.

Of course that's read-only. For editing, it's git clone, edit, push.

In the fullness of time, support for "//server/.." type windows share should of course also work.
It's probably a small actual code change, since the file browser sees them fine, and the 'home' display also opens, but then the code that tries to handle all the options for relative paths fails.

kasemir, 2021-01-06:
I see your screenshots, but I don't see how it gets into that situation.

Trying to open `/als/public/beamlines/...`, it fails with an error for `C:\als\public\beamlines\...`.
I guess that's when it tries to convert that path into an absolute local file system path, erroneously assuming it's `C:\...`.

What's the path that should work?  `file:////als/public/beamlines/...`?

Do debug/fix, you'd have to set breakpoints in https://github.com/ControlSystemStudio/phoebus/blob/master/app/display/model/src/main/java/org/csstudio/display/builder/model/util/ModelResourceUtil.java, and/or add an example to the associated unit test, https://github.com/ControlSystemStudio/phoebus/blob/master/app/display/model/src/test/java/org/csstudio/display/builder/PathTest.java

tynanford, 2021-01-06:
The web server idea is interesting and sounds like a much better approach. The edit feature probably isn't too important but I noticed it while trying to debug this.

The full path would be file:////als-filer/als/public/beamlines/...

So the home display set up looks like this:
org.phoebus.ui/home_display=file:////als-filer/als/public/beamlines/launcher.bob

I will start to debug this later and see if I come up with anything. Thanks for the assistance.

tynanford, 2021-01-07:
I found that opening in a new tab brings up an error, but opening and replacing the current display works fine. 

The relevant code seems to be this if block:

https://github.com/ControlSystemStudio/phoebus/blob/1aaa77d2adb4d482a89ce8cc6f50be3a933a8fc2/app/display/runtime/src/main/java/org/csstudio/display/builder/runtime/ActionUtil.java#L108

kasemir, 2021-01-07:
Don't understand. At that point, the `new_model` has already been loaded, so how can there be a problem resolving the display at that point? Or is the `new_model` displayed alright, and problems then start with embedded content inside that `new_model` which resolves when `new_model` is in the existing tab, not in a new one?

tynanford, 2021-01-07:
It's an interesting problem. I can open the file from file browser, but when I close and re-open Phoebus, I get an error (simply from it trying to open the last state saved in the memento file).

`
java.lang.Exception: Cannot load '/als/public/tynan_test/main.bob' (parent: 'null')
...
Caused by: java.io.FileNotFoundException: C:\als\public\tynan_test\main.bob (The system cannot find the path specified)
`

Anyways I will keep looking but I am mostly just setting breakpoints and stepping through. Guess this will help me learn more about how Phoebus works

kasemir, 2021-01-07:
> when I close and re-open Phoebus, I get an error 

The display files are meant to work on Windows, Linux, Mac.
If somebody creates a file on Windows that embeds another display found in `sub\sub\other.bob`, that type of windows path would not be recognized on Linux/Mac.
So file paths for related displays etc. are at some point run through the `ModelResourceUtil.normalize()` method which replaces backslashes etc.

It might be the code that erroneously removes the `server` from `\\server\xxx`, and when the result is then stored as the memento, it no longer loads.

Again try to find what's erroneously turned into what else, then we can add a unit test to PathTest.java and fix it.

tynanford, 2021-01-07:
Ok sorry this is mostly to document what I have found so far. 

At this line, the variable `info`, of type DisplayInfo, looks wrong

https://github.com/ControlSystemStudio/phoebus/blob/1aaa77d2adb4d482a89ce8cc6f50be3a933a8fc2/app/display/runtime/src/main/java/org/csstudio/display/builder/runtime/app/DisplayRuntimeInstance.java#L270

DisplayInfo info.path = /als/public/beamlines/EPSPhoebus/home.bob

This eventually leads to `resolveResource` and get absolute path, where the C: gets added. But I think the problem is that "file" in `file.getAbsolutePath()` is already wrong.

https://github.com/ControlSystemStudio/phoebus/blob/1aaa77d2adb4d482a89ce8cc6f50be3a933a8fc2/app/display/model/src/main/java/org/csstudio/display/builder/model/persist/ModelLoader.java#L46

https://github.com/ControlSystemStudio/phoebus/blob/1aaa77d2adb4d482a89ce8cc6f50be3a933a8fc2/app/display/model/src/main/java/org/csstudio/display/builder/model/util/ModelResourceUtil.java#L309

This takes us all the way back to the below code.

https://github.com/ControlSystemStudio/phoebus/blob/1aaa77d2adb4d482a89ce8cc6f50be3a933a8fc2/app/display/runtime/src/main/java/org/csstudio/display/builder/runtime/app/DisplayRuntimeApplication.java#L70

Variable values **when I get an error.** (open from action button)

resource=file://als-filer/als/public/beamlines/EPSPhoebus/home.bob  
info.path=/als/public/beamlines/EPSPhoebus/S01/BL01_FE_EPS.bob  
info.resolve=true
input.path=/als/public/beamlines/EPSPhoebus/home.bob
input.string=file:/als/public/beamlines/EPSPhoebus/home.bob

Variable values **when things go right** (when I open from file browser)

resource=file:////als-filer/als/public/tynan_test/main.bob
info.path=//als-filer/als/public/tynan_test/main.bob
info.resolve=true
input.path=/als/public/tynan_test/main.bob
input.string=file://als-filer/als/public/tynan_test/main.bob

tynanford, 2021-01-07:
Well sorry maybe this was pointless. `resource` is "wrong" in my example because the memento file saves the resource path that way.

kasemir, 2021-01-07:
The only part that's obvious to me is that the memento contain the wrong value.
Question is how it gets there.
Should be somewhat like this:

1) You open the file from file browser or 'home' screen.
2) Some parts of ModelResourceUtil are called to locate the screen, which succeeds!
3) The 'normalized' path is kept with the display info, you should see it via the top entry in the context menu that you get from the 'tab' of the display.
4) On shutdown, that path from the display info is written to the memento

Some step after 3 must remove the `//server/..` from your path.

tynanford, 2021-01-07:
Just to be clear, the memento file does contain the server name. The only difference seems to be the number of slashes preceding the server name (2 vs. 4)

Memento:
`input_uri="file://als-filer/als/public/beamlines/EPSPhoebus/home.bob`

This is the line removing the server name:
https://github.com/ControlSystemStudio/phoebus/blob/1aaa77d2adb4d482a89ce8cc6f50be3a933a8fc2/app/display/runtime/src/main/java/org/csstudio/display/builder/runtime/app/DisplayRuntimeApplication.java#L70

It seems like the action button failing might be a separate issue.

tynanford, 2021-01-07:
I can confirm that changing memento file manually to `input_uri="file:////als-filer/als/public/beamlines/EPSPhoebus/home.bob` works. No errors. 

This seems to make sense based on the other things I see working, the home display and file browser. From my settings.ini file:

`org.phoebus.ui/home_display=file:////als-filer/als/public/beamlines/home.bob`  
`org.phoebus.applications.filebrowser/default_root=\\\\als-filer\\als\\public\\beamlines\\EPSPhoebus`

I got those working by looking at things like this: https://stackoverflow.com/questions/41926148/java-how-to-access-unc-location

The 4 slashes seem important for Java to recognize it is not a normal file location.

closed 2023-05-25 by tynanford