ControlSystemStudio/phoebus #1730: Alarm Config Logger does not handle comma separated topics

jacquelinegarrahan, 2021-01-12:
A comma separated list of topics passed via the -topics argument is treated as a single topic:

[image]

[image]

georgweiss, 2021-01-13:
This is indeed a bug as the alarm config logger uses colon (:) as separator.
Will fix this.

georgweiss, 2021-01-14:
Fix in PR #1732.
Can we close this?

shroffk, 2021-02-01:
@jacquelinegarrahan can this be closed it looks like the fix has been merged

jacquelinegarrahan, 2021-02-01:
@shroffk sorry for the delay

closed 2021-02-01 by jacquelinegarrahan