ControlSystemStudio/phoebus #1762: Channel table/tree ~size input persists on separate queries

tynanford, 2021-02-03:
I have been testing out now limiting the size of channel table and channel tree queries with ~size. Kind in reference to this https://github.com/ControlSystemStudio/phoebus/issues/1549

Some unexpected behavior is when I search 

1. * ~size=10 (get 10 results)
2. * (get 10 results)

I also notice this persists between channel table and channel tree. i.e. if I limit a channel table query to size=10, then open channel table and search *, the results are limited by 10.

I was expecting that on *, I would once again see the 10,000+ channels. Is this right?

tynanford, 2021-02-03:
Well, now I am testing with curl and I see the same results. So it looks like this is then either a channelfinder or elasticsearch feature/bug.

shroffk, 2021-02-03:
Ok, I think this is a channelfinder service bug which was introduced with the latest change to make the max size configurable instead of hard coded to 10000
Looking into it

shroffk, 2021-02-03:
https://github.com/ChannelFinder/ChannelFinder-SpringBoot/pull/27

tynanford, 2021-02-03:
Sure thanks! I will let you know how it goes.

Slightly off topic to this issue, are you using the max query param? I couldn't seem to get that to work this morning on our CF instance. Still need to do more debugging on my end to see what might be happening, but wanted to see if you are using it.

tynanford, 2021-02-03:
Confirmed it is working, looks good to merge. Thanks

closed 2021-02-04 by tynanford

tynanford, 2021-02-04:
Just a note on the max query param in CF. To get a different size than 10,000 to work, I had to do this: 

`curl -XPUT "http://localhost:9200/cf_tags/_settings" -d '{ "index" : { "max_result_window" : 500000 } }' -H "Content-Type: application/json"`
`curl -XPUT "http://localhost:9200/cf_properties/_settings" -d '{ "index" : { "max_result_window" : 500000 } }' -H "Content-Type: application/json"`
`curl -XPUT "http://localhost:9200/channelfinder/_settings" -d '{ "index" : { "max_result_window" : 500000 } }' -H "Content-Type: application/json"`

https://stackoverflow.com/questions/41677198/result-window-is-too-large-from-size-must-be-less-than-or-equal-to-10000-b/41677310

shroffk, 2021-02-04:
Yes, 
The initial 10000 limit was in place due to elastic

shroffk, 2022-02-25:
@tynanford 
Should the "max_result_window" be only set during the creation of the index or when ever Channelfinder is restarted the new setting should override the old one