ControlSystemStudio/phoebus #1763: Mac OS 'Big Sur' issues

georgweiss, 2021-02-04:
Seems Mac OS 'Big Sur' has font issues with older Java 11 releases. Upgrading to 11.10+9 seems do the trick.

harley7991, 2021-03-31:
Font issues persist with Mac OS 11.2.3, Java 14, and Phoebus 4.6.6

[image]

[image]

kasemir, 2021-03-31:
I see similar issues with the older Mac OS 10 Catalina, not on the laptop display, but on an external 'thunderbolt' display.
The older Mac laptop had thunderbolt connectors. The current laptop only has USB-C, so a USB-C to thunderbolt adapter connects to the external monitor. But in that case it's a general issue, not specific to any particular application.

In your case, is there any difference when adding `-Dprism.lcdtext=false` or `-Dprism.lcdtext=true` to the start options?

kasemir, 2021-03-31:
What actually helps with my external monitor issue:

Under system preferences, general, de-select "Use font smoothing when available".
From the online help:
`When text smoothing (or “antialiasing”) is on, smaller fonts may be harder to read.`

[image]