ControlSystemStudio/phoebus #1771: Ellipse widget line_width

tanviash, 2021-02-17:
When a legacy ellipse widget with a non-zero line_width is opened in Display Builder, its line_width is seen as 0 and it becomes invisible.

kasemir, 2021-02-17:
Likely same issue as with the 'arc' widget, #1769

@shroffk updated the 'static' widgets to import the legacy 'border' as the 'line', overwriting the actual line width and color.

Fix for the ellipse should be the same as for the arc, just remove that code which replaces the line that you want with some legacy border that's not used, except Kunal might have some example where the border should take preference??

tanviash, 2021-02-17:
@kasemir Thanks, I'll submit a PR soon.

kasemir, 2021-02-17:
For the ellipse, it should be safe to just remove the 'CustomWidgetConfigurator'.
The legacy border would create a rectangular outline, clearly wrong for an ellipse, i.e. shouldn't replace the line color/size/type of the ellipse. But maybe Kunal has a counterexample.

shroffk, 2021-02-18:
As the issue #1564 describes the changes I made were to address the overlap between border and outline. I think we primarily cared about the box so you can go ahead and remove that handling of the legacy border for the ellipse.

Additionally I will add a check in `OutlineSupport.handleLegacyBorder` where if the border width was 0 then the it would be ignored

kasemir, 2021-02-18:
Yes, the box/rectangle certainly worked out fine, maybe the hunch to simply apply that to all graphics wasn't quite right.

shroffk, 2021-02-18:
Sure, @tanviash you can go ahead and remove the handle legacy border for the arch, ellipse, and polygon.

closed 2021-02-19 by tanviash