ControlSystemStudio/phoebus #1774: Databrowser: org.csstudio.trends.databrowser3.Messages contains superflous messages

pavel-ch, 2021-02-22:
I see these messages on startup:
```
2021-02-22 22:05:07 SEVERE [org.phoebus.framework.nls.NLS] org.csstudio.trends.databrowser3.Messages contains superflous message 'ActionEmailTitle'
2021-02-22 22:05:07 SEVERE [org.phoebus.framework.nls.NLS] org.csstudio.trends.databrowser3.Messages contains superflous message 'ActionEmailBody'
2021-02-22 22:05:07 SEVERE [org.phoebus.framework.nls.NLS] org.csstudio.trends.databrowser3.Messages contains superflous message 'ActionLogbookBody'
2021-02-22 22:05:07 SEVERE [org.phoebus.framework.nls.NLS] org.csstudio.trends.databrowser3.Messages contains superflous message 'ActionLogbookTitle'
2021-02-22 22:05:07 SEVERE [org.phoebus.framework.nls.NLS] org.csstudio.trends.databrowser3.Messages contains superflous message 'NewPlotTargetFileNotDirectory'

```
Is that something not-yet-deleted, or a spare?

kasemir, 2021-02-22:
I think these are just obsolete messages that can be deleted.

shroffk, 2021-02-23:
These messages were used by the hard coded email and log entry dialog which was replaced by the one contributed via context menu. We should be able to remove these

shroffk, 2021-03-10:
@kasemir it seems like in the Messages we have a few dozen more NLS messages which are not used anywhere. Should I clean them up too?

shroffk, 2021-03-10:
#1822 addresses this issue.

The startup should not have any warnings now

Additionally, a set of messages were removed which were unused ( they were previously used by the eclipse preferences pages, the logbook/email dialogs, and other unused dialogs)

closed 2021-03-10 by shroffk