ControlSystemStudio/phoebus #1786: Please update README.md regarding Eclipse builds

MichaelRitzert, 2021-03-03:
Each time, I pull from master, my build in Eclipse is broken and I have a hard time fixing it. Usually I end up performing a wild sequence of adding missing .project files, refreshing the Maven project, reinstalling lombok, cleaning, building with mvn on the command line, updating the run configuration, …, until phoebus starts. Only to find that only one or two modules are available with the configuration I ended up with…

It would be really helpful if README.md contained up-to-date information on how to get a reliable build system with Eclipse (or a clear recommendation to use another IDE).

kasemir, 2021-03-03:
Building with maven is listed first, building with ant second, and I think those are very reliable.

Use of an IDE is a personal matter. We can move the Eclipse instructions "down" one level behind Intellij or VSC or ..
Having said that, I cannot reproduce your issue when I use the "plain Java configuration" approach, importing those projects that have a .project and .classpath file.
You seem to use the Maven Files in Eclipse, on which I cannot comment.

MichaelRitzert, 2021-03-03:
There is nothing in the file that says that the options are listed in order of preference, so I thought that all of them are considered equal. (Or am I looking at the wrong/old file since you mention VSC which I cannot find: https://github.com/ControlSystemStudio/phoebus/blob/master/README.md ? That one looks really outdated and I already had my doubts about it.)

I simply prefer using an IDE because the compile-as-you-type approach gives me a much faster turnaround time compared to waiting for a full mvn build (which I can confirm works very reliably) to finish after a tiny change.

kasemir, 2021-03-03:
You're looking at the correct file.
It was last updated 3 month ago.
For Eclipse, it describes how to "Use plain Java configuration", Use File, Import, ... That's what I use and it's been OK so far.
Kunal used the approach listed next where you add the "Maven Integration for .. Eclipse". I do remember that it wasn't perfect. Eclipse basically creates .project and .classpath files from the maven config, not always doing that well, so my decision was to simply use the existing .project and .classpath files when developing in Eclipse, but maybe that's by now better. Kunal would know more.
Others use Intellij or NetBeans, both I think based on the maven files. You can use VSCode and basically run maven from within that. We haven't had anybody clamor for gradle, yet.

shroffk, 2021-03-03:
Hi Michael,
I understand your frustration with eclipse. 
With the old eclipse based cs-studio we used to run into similar issues of random error which disappear when the project was refreshed enough number of times.... additionally we had the same situation where errors were rarely reproducible. 

An IDE is indispensable for development, however it is difficult to reproduce and address most of the m2eclipse based issues.  While I do use the eclipse IDE as described in the readme. I am migrating to IntelliJ (which needs learning but is VERY consistent).

MichaelRitzert, 2021-03-03:
I thought I'd use the maven-in-eclipse approach because that's what I know from CSS. I just tried "plain Java", and it failed with display-navigation not being a project. What would be the correct (UI) way to fix this instead of just putting a .project file there as I did? After that small obstacle, I have to say it was the easiest solution I've seen so far.

kasemir, 2021-03-03:
Since I'm not actively developing the display-navigation, I've not created plain .project and .classpath files for them.
If you want to, it's usually an easy thing to do based on copy/pasting existing ones and editing the name, maybe the dependencies, then please add them to a PR.

Or check out the other IDEs which are better at using maven.

shroffk, 2021-03-03:
Sorry, 
I had not added .project or .classpath to the display-navigation file (and maybe a few other projects) .
I will add them

georgweiss, 2021-03-04:
At ESS we're using the free community version of IntelliJ. It is not without minor pain points, but I find the IDE easy to work with and it comes with a number of small features that makes the development process smoother compared to Eclipse.

closed 2023-08-15 by shroffk