ControlSystemStudio/phoebus #1787: Expose DESC field

georgweiss, 2021-03-04:
We have a request to support access to the DESC field of a record. The purpose would be to be able to show it in a tooltip.
Based on my knowledge of EPICS I've tried to analyze the impact and need community input to determine how to proceed with this request.

In the application layer we are basically operating on VType classes, so the value of the DESC field would have to be added there, e.g. in the Display class or as a new String field. For ca I assume this would require a change in jca. In pva the DESC field is decoded, so this would make it easier to meet the request.

I would appreciate input from EPICS veterans on this topic, and maybe some pro/contra arguments.

kasemir, 2021-03-04:
If you do have records, you can read "NameOfRecord.DESC" as a separate channel. But you cannot make the "DESC" information part of the metadata of every numeric channel.

This request misses the basic difference between records and channel access.
Just like authors of a news website and http.
Authors fill the web site, and you can read the site via http. Doesn't mean you can get the sock color of each author.
Records fill channels, but that doesn't mean you can get every aspect of the record via channel access.
The authors can be replaced with AI robots. The records can be replaces with python, labview, Matlab code. The result is still accessible via http resp. channel access, but there is no sock color.

For more examples see "Channel Access" on https://controlssoftware.sns.ornl.gov/training/2019_USPAS/

georgweiss, 2021-03-04:
Not sure I understand completely. 
Let's compare with the EGU field of a record. It is available as a field in the Display class, so the data on the wire is decoded such that the field can be set, if the record defines EGU.

kasemir, 2021-03-04:
Yes, the EGU field of records turns into the units property of the channel.
But the numeric properties are limited to units, precision, display/control limits.
There is no description property (filled from the DESC field),
there is no read back property (filled from a motor records RBV field),
there is no hysteresis property (filled from an analog records HYST field),
there is generally not a property for every field.

This abstraction from records and fields to channels and properties allows the decoupled development of servers and clients. Servers can be IOCs with records, but don't have to be.
If you want to add a description property to the channel, you'd have to fundamentally change channel access, i.e. bring that up in the core developers group.

georgweiss, 2021-03-04:
I realize ca is out of scope for this request.
In pva however the PVAStructure class decodes the DESC field so it is accessible and could be used to set a field in the Display class. Right?

pavel-ch, 2021-03-04:
I would say, with ca, you can use a script, bind to the $(pv_name).DESC and modify (append) the Tooltip property "specification" text on its value, expecting only single script activation per widget lifetime.
With pva that would change to $(pv_name)/DESC, right?

kasemir, 2021-03-04:
Here's the "pvinfo" output for a calc record served by an EPICS 7 IOC:
```
    epics:nt/NTScalar:1.0
        double value
        alarm_t alarm
            int severity
            int status
            string message
        time_t timeStamp
            long secondsPastEpoch
            int nanoseconds
            int userTag
        structure display
            double limitLow
            double limitHigh
            string description
            string units
            int precision
            enum_t form
                int index
                string[] choices
        control_t control
            double limitLow
            double limitHigh
            double minStep
        valueAlarm_t valueAlarm
            boolean active
            double lowAlarmLimit
            double lowWarningLimit
            double highWarningLimit
            double highAlarmLimit
            int lowAlarmSeverity
            int lowWarningSeverity
            int highWarningSeverity
            int highAlarmSeverity
            byte hysteresis
```
Yes, there's a 'description' in there, and it's filled from the DESC field of for example a CALC record.

georgweiss, 2021-03-04:
Thanks @pavel-ch for the suggestion. However, we discourage our users from using scripts, even if it is only at instantiation time.

kasemir, 2021-03-05:
> .. VType .. value of the DESC field would have to be added there, e.g. in the Display class or as a new String field.

Well yes, a description field could be added to the VType `Display` as used by numeric types `VDouble` etc, and also to the `VString`.

The PV Access client could indeed populate it from the `display/description` of a received `NTScalar`. When talking to records on IOCs, `qsrv` will use the DESC field of records to fill the `display/description`. Other PVAccess servers written in python or Java or .. can put whatever they want in there, again we have the separation of records&fields from channels&properties in normative types.

For  Channel Access, there's no good solution. Updating the underlying `dbr_control_xx` data types that are exchanged over the network is pretty much out of the question. The phoebus PV layer could establish a separate channel, i.e. for each PV "ABC" it would also attempt to create a channel "ABC.DESC".  But that basically doubles the number of channels, i.e. doubles the amount of memory used on each of the connected CA servers, i.e. IOCs. With the growing number of CA servers that are not IOCs, it'll result in disconnected channels, i.e. brief floods and then a lower but constant base traffic of unfulfilled name search requests on the network. So for channel access I think the VType will have an empty description.

shroffk, 2021-03-05:
I think expanding the functionality for PVA is great but for CA I think we should acknowledge that this is a limitation of the protocol and the above point shows that we might want

> The phoebus PV layer could establish a separate channel, i.e. for each PV "ABC" it would also attempt to create a channel "ABC.DESC". But that basically doubles the number of channels, i.e. doubles [...]

A very strong +1 on this.. for CA the description field should be empty... if you want the description then move to PVA.

pavel-ch, 2021-03-05:
Please take into account that other protocols may be in use, like MQTT, ... The final solution should be protocol neutral, should that be for example a "by default" empty description field, optionally filled on protocol layer.

kasemir, 2021-03-05:
Correct: Just like `ca://`, the `sim://`, `loc://`, `mqtt://`, `sys://` type PVs will leave the description empty.
With `VTypes` basically being the common denominator, it makes you wonder if just `pva://` should drive the expansion.

Still, it is sometimes nice to have the 'description'. So what do you do?
On displays, you can simply add a text update for "XXX.DESC".
The PV Table does have a Description column, it does establish a separate channel/PV, and it does fail to connect for non-IOC PVs. So there's a preference setting to disable it.
Similarly, the PV Tree fails to read "XXX.RTYP" for non-IOC PVs.
So we have examples for specific tools which hack around the limitations of Channel Access, they do establish extra channels and/or fail to connect. But that behavior is limited to just running the tools, it's an accepted evil.
I would worry to have every Channel Access PV do this.

georgweiss, 2021-03-05:
Thanks all for the feedback.
The original request was to be able to show the DESC field in the tooltip of a widget, hence my proposal to add this string somewhere in the VType class hierarchy. Would suggest a null value (i.e. not empty string) if not set.
Agree that this would be a pva feature only.

shroffk, 2022-08-10:
@georgweiss 
The new release of epics-vtypes is now published.
I have updated the dependencies in Phoebus.

what needs to be updated to complete this Issue

georgweiss, 2022-08-11:
The request was essentially to expose DESC as a built-in symbol, e.g. pv_desc. With ca I guess this would have to be the empty string.

Will create a PR soon.

closed 2022-08-15 by georgweiss