ControlSystemStudio/phoebus #1836: Traces not loading intermittently in databrowser

rjwills28, 2021-03-24:
**Issue summary**
We have come across an issue with the databrowser in both CS-Studio and Phoebus, when requesting and plotting archived data for multiple PVs from the Appliance Archiver. Sometimes the full trace for one of the PVs is not plotted for the selected timeframe - it is simply cut off. On other times the plot does show the full trace. It is very intermittent. In the particular example that highlighted this issue there were 21 AA requests being made when the plot was loaded from file. 

**Issue**
I believe the issue here is triggered because this plot contains so many AA requests (all on separate threads). I have had a look into this and here is a detailed description of what I think is happening in the code:

There are 21 AA requests in this plot. Each PV is queried on it's own `WorkerThread`, which uses an `ApplianceArchiverReader` object to construct and send the request. The data is returned to the reader in the form of an iterator object and this object is stored internally in a synchronized map held by the reader. It is also returned to `WorkerThread` which then iterates over the iterator obtaining the actual data. Following this, the `WorkerThread` makes a call to close the reader. Within the reader, the `close()` method calls `cancel()` and here is the main issue.
So while this thread is running on this PV (call it PV1), another thread is running the request for another PV (call it PV2), but it uses the same reader. The returned iterator for PV2 is added to the synchronized map and this `WorkerThread` starts to iterate through the data. In the meantime, PV1 has finished so will be removed from the map and then call the `close()` -> `cancel()` methods in the reader. The `cancel()` method goes through each PV in the synchronized map (which will still contain PV2) and closes it. This means that PV2 gets closed before all of the data has been read from it and hence we end up with an incomplete trace being plotted.
This issue is not always seen as it all depends on when data is read from the iterator and when the iterator gets removed.

Why does this happen:
First thing I noticed is that a 'new'/ unique `ApplianceArchiveReader` object is not used for each PV request. Instead the object is looked up from the cache. If it does not exist then a new one is created and added to the cache. As a tests, if I ignore the cache and create a new reader for each PV then it behaves correctly, however this takes more time to create a new object each time.

So assuming using the cached object is correct (i.e. that we should be using the same reader), then the `close()` method should not call `cancel()` as it is the `cancel()` method that removes all of the PV iterator objects even if they are not complete. Each PV iterator will be removed from the mapping when the read has finished anyway so at the end of all the queries the map will be empty. I do think the `cancel()` method is implemented correctly though as if the look up is cancelled, then the mapped PVs should be removed to release resources.

**Suggested fix?**
A quick fix that I have tested is simply to have the `close()` method in the `ApplianceArchiveReader` do nothing. I can not see any immediate side-effects of this and I am convinced that the iterator map will be emptied naturally as the requests complete. That said, I am still a little cautious about this as the fix. Was it intended that same reader should be used each time - not sure as then calling `close()` on it each time seems incorrect? @kasemir do you have any thoughts on this?

rjwills28, 2021-03-24:
Example: 
Below are the complete traces:
[image]

On another load, the full trace (cyan) was not plotted:
[image]

kasemir, 2021-03-24:
> Each PV is queried on it's own WorkerThread, which uses an ApplianceArchiverReader ...

Yes, the data browser starts several retrieval jobs in parallel.
In Phoebus, this can be adjusted via the `org.csstudio.trends.databrowser3/concurrent_requests` settings.
Which exact reader is used, be it `ApplianceArchiverReader` or `RDBArchiveReader` depends on the archive URL.

Whatever the underlying data source archive appliance or RDB, must support more than one client because after all you may have more than one person look at archived data at a time.
So in principle just one person running just one instance of the data browser should also be able to make more than one connection in parallel to speed up fetching data for N channels, but in reality there are likely limits. The data source only allows like 100 parallel connections, and while you might see a performance gain when running 5 parallel sessions, beyond that there might not be any gain. So you can try to tune that value for your site.

> this object is stored internally in a synchronized map held by the reader.

This may be true for the `ApplianceArchiverReader`. The `RDBArchiveReader`  internally caches RDB connections, but not the reader itself.

> .. WorkerThread makes a call to close the reader... So while this thread is running on this PV (call it PV1), another thread is running the request for another PV (call it PV2), but it uses the same reader...

That looks like a bug in the `ApplianceArchiverReader`. It's perfectly fine for it to use for example a map to cache instances for re-use, but it must obviously only re-use them once they're done, i.e. when the worker thread 'closes' the reader. Can't run concurrent requests for both PV1 and PV2 with the same reader.

> A quick fix that I have tested is simply to have the close() method in the ApplianceArchiveReader do nothing. 

While that might work around an issue in the internal pooling/caching of the `ApplianceArchiveReader` right now, from what you describe it would still result concurrent use of the same reader for PV1 and PV2. Plus you likely can't leave the reader instances 'open' in the 'map' forever, eventually the network connection will time out, so then you have a reader in the map that's 'open' but will fail as soon as its used again.

kasemir, 2021-03-24:
> A quick fix that I have tested is simply to have the close() method in the ApplianceArchiveReader do nothing.

An equally quick fix might be to simply remove the map.
Archive readers get created, used, then closed.
If some underlying mechanism wants to pool the connections, that is probably better handled by a connection pool layer:
New ApplianceArchiverReader checks pool to create or re-use a connection, uses it, returns it to the pool on `close()`.

rjwills28, 2021-03-24:
Thanks for the detailed feedback.

It is the `ApplianceArchiveReaderFactory` class that is creating a single instance of the `ApplianceArchiveReader` and then adding this to a map to cache it simply based on the AA reader url. It then re-uses this cached `ApplianceArchiveReader` each time which means we get concurrent calls into it. 
So would you suggest that we just remove the map in `ApplianceArchiveReaderFactory` so that it just creates a new `ApplianceArchiveReader` and returns that each time?

kasemir, 2021-03-24:
Yes, skipping the map should be the easiest way to fix the problem you  described.
So first, make it "work".

Next step could then be to figure out what can be re-used and how.
Like the HTTP connection might be re-usable, as long as only one ApplianceArchiveReader is using it at a time.
So I would probably still create and close the ApplianceArchiveReader instances, but they may share HTTP connections.

shroffk, 2021-07-09:
I am having trouble testing this myself... I have copied your patch from cs-studio.