ControlSystemStudio/phoebus #1842: Timestamps?

thomasms, 2021-03-29:
I have a simple pva server running which every second generates some random value, the PV looks like:
```
structure 
    float x 7.07089
    string name default:component:1
    float mean 10
    float sig 2
    alarm_t alarm 
        int severity 0
        int status 0
        string message "No problem"
    time_t time 2021-03-29 15:16:30.844  -1 
        long secondsPastEpoch 1617023790
        int nanoseconds 843611000
        int userTag -1
```
There is no problem with this server and by using `pvget` I indeed see the timestamp updating.

However, when I hook a PV up to a widget in Phoebus (using Display Builder), the timestamp does not propagate to the tooltip, as you can see in the screenshot below. (my PV is named 33358fbf-875d-4417-a33d-458e245d3694).

[image]

I have tried naming it `time`, `timeStamp`, and `timestamp` but none work, is this due to the structure of my PV? 
Should I instead have a timestamp nested under each field?

It should be noted that if I create a text entry field for the time itself (pva://..../time) it does indeed show in the tooltip for that widget, it is just somewhat confusing to a user if the timestamp for another field is not correct. 
It may be useful to just disable it in the tooltip instead, any guidance on how to do so would be greatly appreciated.

kasemir, 2021-03-29:
When you have a plain IOC, for example using 'softIocPVA', you can read/write those PVs via channel access and PVAccess.
Those use 'normative types', where the time stamp is in a "timeStamp" element and so on as defined in the normative types.

If you create your own custom structure, you cannot expect generic clients to guess that "time" or "ZeitStempel" whatever you call is is supposed to be _the_ time stamp. If you rename your time stamp from "time" to the normative name "timeStamp" you may have more luck, but in general custom structures require both a custom server to create them (you did that) and a custom client to digest them.
Yes, the "timeStamp" has to be below the data element.

thomasms, 2021-03-29:
No indeed, I fully understand, there must be a contract between the client and server to know what to expect. It's just I did the same for the alarm and it picked that up OK, so I assumed (maybe this was my mistake) that the same would work for the timestamp. 

I did also try "timeStamp" as a name but it did not work, unfortunately.  

Are you saying if I did something like this:
```
structure 
    float x 7.07089
    alarm_t alarm 
        int severity 0
        int status 0
        string message "No problem"
    time_t timeStamp 2021-03-29 15:16:30.844  -1 
        long secondsPastEpoch 1617023790
        int nanoseconds 843611000
        int userTag -1
```
it would work? Since it is a NTScalar.

kasemir, 2021-03-29:
I think the issue is related to your structure not having a `value`, and then the `timeStamp` would be located relative to the `value`.

You'll have to debug https://github.com/ControlSystemStudio/phoebus/tree/master/core/pv/src/main/java/org/phoebus/pv/pva, most likely the code around  `decodeTime()`,
https://github.com/ControlSystemStudio/phoebus/blob/70b9a4c3a7feefc7cf86c97f328c443af8468fc5/core/pv/src/main/java/org/phoebus/pv/pva/Decoders.java#L140
to see why it's not using that.