ControlSystemStudio/phoebus #1847: Image widget has the ROI Y-axis reversed

domitto, 2021-03-31:
I've configured the Overlay Plugin (blue rectangle) to follow the ROI plugin but the widget shows the ROI with the Y-axis reversed (red rectangle):

[image]

domitto, 2021-03-31:
Same PV with cs-studio (eclipse based) and intensity graph:
[image]

kasemir, 2021-03-31:
When you say "Overlay Plugin", I assume you imply that you're using the area detector.

I think the area detector always uses a coordinate system where Y=0 is on top, and Y grows towards the bottom.
In the image widget, you can configure the Y axis to be whatever you want.
If the image has N rows, you can configure it as minimum=0 and maximum=(N-1).
Or minimum=(N-1) and maximum=0.
Or minimum=-1.0 and maximum=+1.0.
Or minimum=0.0 and maximum=+1.0.

The area detector doesn't care how you configure the image widget, and the image widget doesn't care what coordinate system the area detector uses.

If you want them to match, you need to configure the image widget to match the area detector coordinate system, because you can configure the former yet not the latter.
This likely means:  Y axis minimum=(N-1) and maximum=0.

domitto, 2021-04-04:
Hi Kay, thanks for the explanation. I was able to change the Y 0 to be at the top of the image and it fixed the issue. 
I never had to do that with cs-studio utilizing the intensity graph and it was a little confusing seeing the ROI with the Y-axis reversed.

Thanks again,
Diego

closed 2021-04-04 by domitto