ControlSystemStudio/phoebus #1848: Data browser trace issues

kasemir, 2021-03-31:
Add trace "A", then trace "B" -> You have traces "A", "B".
Delete trace "A". Undo. -> You now have traces "B", "A". Correct color, same axis assignments, but order in properties has changed.

Add trace "A", then trace "B". Add annotation to trace "A". Hide trace "A".
Add trace "C". Show trace "A" again -> Annotation on trace "A" is gone.

The two issues are likely related. Adding/removing traces can change the order of traces. Annotations are associated with traces via the index 0, 1, .. of the trace in the list of traces, and as that order changes, the association is lost.

Suggested fix is to internally hold annotations as members of the trace. This change can be limited to the data browser model, not affecting the *.plt file nor the RTPlot.

thelarsjohansson, 2024-04-11:
@kasemir 
Hi!

Similar / same behavior has been found when .plt file is being opened by Phoebus, There is IndexOutOfBoundsException since 

  <annotations>
    <annotation>
      <pv>-1</pv>

pv is referred to with index -1. It's not yet clear as to how files became like this but it may be for reasons mentioned in text above.

What's the status of this issue?

Kind regards
Lars Johansson (ESS)

kasemir, 2024-04-11:
Haven't found time to look into it, un-assigned myself since I won't get to it in the near future

thelarsjohansson, 2024-04-25:
It might be multiple issues behind behavior discussed here

- delete of trace with annotation does not adjust item index for remaining annotations
      remaining annotations might refer to wrong or non-existing pv

- undo, after delete of trace with annotation, does not bring back annotation

- annotation item index with value -1
      -1 is not explicitly set but might come from indexOf operation, to find index for pv, returning -1 for index not found

thelarsjohansson, 2024-04-25:
I am currently testing solution candidate for 1st issue.

It may be that 3rd issue mentioned, annotation item index -1, is related to 1st issue. 

If annotation, that is left after delete of other pv with annotation, refers to non-existing pv and annotations then are updated ( e.g. after annotation is moved in view ), then update tries to find non-existing item index for annotation ( list.indexOf() method ) and item index now ends up as -1.

Fix for 1st issue may solve 3rd issue.

thelarsjohansson, 2024-05-08:
Above mentioned 3 issues fixed. Proceeding with hide/show traces with annotations.

thelarsjohansson, 2024-06-11:
I have solution candidate for the remainder of this issue, i.e. annotations related to hiding/showing of traces, as described in first comment in ticket.

Annotations has to be reset for plot when there is change for visibility of item in model.

https://github.com/ControlSystemStudio/phoebus/pull/3039

closed 2024-06-11 by kasemir