ControlSystemStudio/phoebus #1860: Question: option to hide tick marks

georgweiss, 2021-04-14:
My users have a use case where the Meter widget would be useful if the tick marks could be removed. Looking at the current APIs this seems non-trivial for the edit mode case (i.e. toggling tick marks).

Question is if this is a use case we'd like to support. I am prepared to do the changed, but would then in the name of consistency apply that to all cases where a scale/axis applies, i.e. also the plot widget.

kasemir, 2021-04-14:
I'd prefer to keep the meter as a black box. You enter the PV name, maybe set the min/max range, but otherwise you get what you get. There is no way to fine tune the tick placement, length of minor ticks, width of the needle, .. because in 5 years there will be a totally different type of meter and whatever you adjust now won't apply any longer.
Whatever you do for a meter, 50% of users won't like it anyway, so it's more important to have a consistent look on all displays and not end up with each meter looking different based on who last edited that file.
Instead of tweaking the meter in the desktop version, why not spend that time adding one to the web representation, http://github.com/kasemir/dbwr?

georgweiss, 2021-04-14:
Well, the idea was to just have a check box that would say "Show Tick Marks". I'd argue that most users may not even notice it.
In any case, the web representation is not an option for us for the time being, and may not be for quite some time.

Personally I have serious doubts to the use case as presented by my users, and your input is a factor in my dealings with them, so thanks anyway.