ControlSystemStudio/phoebus #1876: Allow site specific products to exclude modules

shroffk, 2021-04-22:
Site specific products should be able to exclude modules that are not needed.

This can be achieved with the used of optional dependencies. The optional dependencies will not be included in site specific products unless they are explicitly redefined.

kasemir, 2021-04-22:
Yes and no. Of course it's good for a site-specific product that needs logbook X to include support for just that logbook, no need to carry support for all the other logbooks around. At the same time, it's nice to have one 'generic' product that includes support for several logbooks, at least in-memory as a standalone demo and olog as an example for a logbook that's not site-specific, and preferences allow selecting which one you want to use.
For archives, we support several clients, and for PVs we support several types, so fundamentally it should be OK to include several logbook clients and simply enable/disable as needed until you create your site-specific product that only includes what you really want.

shroffk, 2021-04-22:
I think I should expand a bit.

I think the common product should have inmemory, olog, maybe even more logbook implementation and users should be allowed to choose. Currently, in the common product I include all the logbook clients and the users can use the preference to select the one they want to you

My point is that no one should be **forced** to include the olog module... and by properly organizing the preferences we can ensure that is not the case.

kasemir, 2021-04-22:
Sounds good.

closed 2021-06-11 by shroffk