ControlSystemStudio/phoebus #1878: managing logbook preferences

shroffk, 2021-04-22:
The modular nature of Phoebus allows the creation of different products with logbook client implementations.

We need to create a set of common/core preferences - like which logbook implementation to use

Currently, these preferences are located in optional app modules.

shroffk, 2021-04-22:
I have create a new preferences under org.phoebus.logbook
This preference in the core module allows the selection of the logbook client implementation.

```
# ------------------------------
# Package org.phoebus.logbook.ui
# ------------------------------

# Site specific log book client implementation name.
# When empty, logbook submissions are disabled
logbook_factory=inmemory
```

kasemir, 2021-04-22:
> I have create a new preferences under org.phoebus.logbook
.. but then the example shows "...ui". Is that a typo in the comment, or is there a problem with the preference name?
Do you want to keep using the original "org.phoebus.logbook.ui/logbook_factory" in case people have already put that into their settings, or should we use a new "org.phoebus.logbook/logbook_factory" setting without UI since it's been moved out of UI?

shroffk, 2021-04-22:
> should we use a new "org.phoebus.logbook/logbook_factory" setting without UI since it's been moved out of UI?

I think we should use the new preference...postponing might just cause confusion in the future

closed 2021-04-22 by shroffk