ControlSystemStudio/phoebus #1885: Handling null `display.units` (and maybe others)

mdavidsaver, 2021-04-26:
I'm working on a new server application using PVXS, and have managed to trigger something strange.  The structure includes `display.units`, but no value is sent.

https://github.com/ControlSystemStudio/phoebus/blob/a37236ac75099a7b7117106d3dc40f0bef98d1a7/core/pv/src/main/java/org/phoebus/pv/pva/Decoders.java#L243

```
2021-04-25 18:33:19 WARNING [org.phoebus.pv] Cannot decode 'simple.md.test!degM' [CID 7, SID 117768965 CONNECTED] = epics:nt/NTScalar:1.0 
    ushort value 317
    alarm_t alarm [NO_ALARM]
        int severity 0
        int status 0
        string message null
    time_t timeStamp [2021-04-25 18:33:19.915067300]
        long secondsPastEpoch 1619400799
        int nanoseconds 915067300
        int userTag 0
    structure display
        ushort limitLow 0
        ushort limitHigh 0
        string description async degree measurement
        string units null
        ubyte ppmSize 0
java.lang.NullPointerException: unit can't be null
        at org.epics.vtype.VType.argumentNotNull(VType.java:190)
        at org.epics.vtype.IDisplay.<init>(IDisplay.java:30)
        at org.epics.vtype.Display.of(Display.java:144)
        at org.phoebus.pv.pva.Decoders.decodeDisplay(Decoders.java:312)
        at org.phoebus.pv.pva.Decoders.decodeShort(Decoders.java:359)
        at org.phoebus.pv.pva.Decoders.decodeNumber(Decoders.java:438)
        at org.phoebus.pv.pva.PVAStructureHelper.decodeScalarField(PVAStructureHelper.java:149)
        at org.phoebus.pv.pva.PVAStructureHelper.decodeScalar(PVAStructureHelper.java:140)
        at org.phoebus.pv.pva.PVAStructureHelper.getVType(PVAStructureHelper.java:94)
        at org.phoebus.pv.pva.PVA_PV.handleMonitor(PVA_PV.java:84)
        at org.epics.pva.client.MonitorRequest.decodeValueUpdate(MonitorRequest.java:236)
        at org.epics.pva.client.MonitorRequest.handleResponse(MonitorRequest.java:198)
        at org.epics.pva.client.MonitorHandler.handleCommand(MonitorHandler.java:38)
        at org.epics.pva.client.MonitorHandler.handleCommand(MonitorHandler.java:18)
        at org.epics.pva.common.CommandHandlers.handleCommand(CommandHandlers.java:66)
        at org.epics.pva.client.ClientTCPHandler.handleApplicationMessage(ClientTCPHandler.java:304)
        at org.epics.pva.common.TCPHandler.handleMessage(TCPHandler.java:383)
        at org.epics.pva.common.TCPHandler.receiver(TCPHandler.java:287)
        at java.base/java.util.concurrent.FutureTask.run(FutureTask.java:264)
        at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1128)
        at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:628)
        at java.base/java.lang.Thread.run(Thread.java:829)
```

I'm working from 70b9a4c3a7feefc7cf86c97f328c443af8468fc5

shroffk, 2021-04-26:
I would prefer to keep the check in VTypes which prevents the units from being null.

I recommend replacing 
` units = str == null ? noDisplay.getUnit() : str.get();`

with

```
            if (str == null || str.get() == null)
                units = noDisplay.getUnit();
            else
                units = str.get();
```
There is still the case where the `units` is an empty string, but that would be the same as a units of noDisplay

kasemir, 2021-04-26:
Yes, check for `str == null || str.get() == null`.

georgweiss, 2021-04-26:
Reminding of #1787. I can update to resolve it,  but we should sync to avoid multiple release procedures.

georgweiss, 2021-04-26:
Sorry, my bad. #1787 is for epicsCoreJava....

kasemir, 2021-11-17:
Closing since the display with null for units is now handled.

closed 2021-11-17 by kasemir