ControlSystemStudio/phoebus #1889: "Show Legend" in 'X/Y plot' doesn't work correctly

haitao127, 2021-04-27:
[image]

By default, the "Show Legend" option is selected, so the X/Y plot is like this:
[image]

If I deselect the "Show Legend" option, what I **expect** should be like this:
[image]

But in **fact**, it will be like the picture below, which shows the 'Legend' displayed on the Y axis:
[image]

kasemir, 2021-04-27:
'Show legend' decides if the trace names are shown in a separate legend or on the value axes.
If you don't want to show a trace name, simply don't configure one, i.e. enter an empty text for the trace name.

haitao127, 2021-04-28:
Ah, I see, thank you.
But I think this option is easy to be misunderstood, it would be better if it could be optimized.