ControlSystemStudio/phoebus #1909: Olog search query adds desc=* after restart

georgweiss, 2021-05-17:
Searching for an Olog entry where body text is empty does not result in any hits unless desc=* is removed (bug? expected behavior?). Removal desc=* from the query string is persisted in the memento file, but is added in code, resulting in the unexpected behavior where the user modified query string is not the same as when Phoebus was shot down.

shroffk, 2021-05-17:
So there are 2 things in play here:
1. I think the elastic query mapped to desc=* expects something to be present. 
match anything != match anything including nothing
If we want to update this behaviour we will have to update the service...I fell this should be resolved on the client side which sends the query.

2. The logbook applications treats empty or null search string as a special case where the default search defined in preferences is used. We could change the default search query preference from desc=* to * or something similar

georgweiss, 2021-05-17:
Is a change in the preference enough, see LogEntryTableViewController:
```
searchParameters = FXCollections.observableHashMap();
        searchParameters.put(Keys.SEARCH, "*");
        searchParameters.put(Keys.STARTTIME, TimeParser.format(java.time.Duration.ofHours(8)));
        searchParameters.put(Keys.ENDTIME, TimeParser.format(java.time.Duration.ZERO));

```

georgweiss, 2021-05-17:
The AdvancedSearchViewController is also a player, I think.

shroffk, 2021-05-18:
The above commit ensures that the table and calendar are initialized via the preferences.

shroffk, 2021-05-19:
@georgweiss can we close this?

closed 2021-05-19 by georgweiss