ControlSystemStudio/phoebus #1917: Annunciator does not work on CentOS 7

georgweiss, 2021-06-01:
We are unable to make the annunciator work on our control room workstations running CentOS 7. The sound system seems to be set up correctly as video media can be played back with sound. 

So I have a question on dependencies... Text-to-speech seems to rely on:
```
<dependency>
      <groupId>net.sf.sociaal</groupId>
      <artifactId>freetts</artifactId>
      <version>1.2.2</version>
    </dependency>
```
Are there any dependencies that the operating system must provide?
What about the Annunciator.py? Is that only for setting up a stand-alone annunciator?

kasemir, 2021-06-01:
Yes, it uses freetts. See https://github.com/ControlSystemStudio/phoebus/blob/master/app/alarm/ui/src/test/java/org/phoebus/applications/alarm/VoiceDemo.java for a standalone demo.
See also https://github.com/ControlSystemStudio/phoebus/issues/1780 and https://github.com/ControlSystemStudio/cs-studio/issues/2591

georgweiss, 2021-06-01:
I understand, but are there any dependencies besides the ones listed in the pom?

kasemir, 2021-06-01:
Should just be the freetts lib and plain JRE. Try `VoiceDemo`.

georgweiss, 2021-06-01:
Thanks, VoiceDemo works fine, as does Annunciator on Mac and Windows. But not on our CentOS control room stations...
Need to dig deeper.

closed 2021-06-01 by georgweiss

kasemir, 2021-06-01:
Does the VoiceDemo work on your CentOS control room stations?

georgweiss, 2021-06-01:
Not tested, but there is no sound when starting the Annunciator table, nor using its test button.
Annunciator works however on a laptop running CentOS, so seems this is related to the control room workstation setup.
Just wanted to confirm that were not missing any dependency.

kasemir, 2021-06-01:
Try the VoiceDemo work on your CentOS control room stations to assert that the basic audio support as used by freetts is functional.

If the standalone VoiceDemo works, but the annunciator doesn't, there could be an event handling conflict similar to what we had way back with the Eclipse-based CS-Studio. The audio system seems to rely on an event handling mechanism related to AWT, and back then that could clash with the SWT event handling. Maybe you're seeing a similar clash between the event handling of the audio system and the JavaFX event queue.

georgweiss, 2021-06-01:
Well, I will probably need to visit the control room sooner or later...

kasemir, 2021-06-01:
If the VoiceDemo works, but the annunciator hangs, look at stack dumps to check for a lock-up between audio and JavaFX. So far, however, the audio problems on Linux have been much simpler, just no Java audio.

georgweiss, 2021-06-03:
In case anyone else should face this issue, here is the command line that solved our problem:
`yum install alsa-plugins-pulseaudio.x86_64`