ControlSystemStudio/phoebus #1921: Alarm: State topic remains when an alarms is deleted

sasaki77, 2021-06-07:
When I delete an alarm from Alarm tree GUI or `-import` command line option, Config Topic is deleted with "tombstone".
However, State Topic remains as it is.
Following is an example of my test.

```json
config:/Accelerator/TEST/Alarm00010: {"user":"sasaki","host":"sasaki-VirtualBox","description":"Alarm00010"}
state:/Accelerator/TEST/Alarm00010: {"severity":"OK","message":"OK","value":"0.0","time":{"seconds":1623029227,"nano":154814561},"current_severity":"OK","current_message":"NO_ALARM"}
config:/Accelerator/TEST/Alarm00010: {"user":"sasaki","host":"sasaki-VirtualBox","delete":"Deleting"}
config:/Accelerator/TEST/Alarm00010: null
```

The alarm server seems to ignore State topics. On the other hand, alarm client seems to create a new alarm tree item from State topic if the item is not created yet.
I know alarm client deletes the alarm from alarm tree when it found "tombstone" record and alarm tree should be consistent eventually while "tombstone" record remain.
But alarm client might have alarm item for deleted alarm until it find "tombstone" record.

I think we can simply ignore State topic for deleted alarm if we also mark State topic as "tombstone".

Are there any reason to keep State topic when we delete an alarm?

kasemir, 2021-06-07:
The `config:` and `state:` messages are sent via the same Kafka topic, so their order is guaranteed.
When an item is deleted, the last message for an item should be the `null` "tombstone" record, similar to your example:
```
# One or more state updates
state:/Accelerator/Top/sim:\/\/sine: {"severity":"MAJOR",...}
# Deletion marker and "tombstone"
config:/Accelerator/Top/sim:\/\/sine: {"user":"...","host":"...","delete":"Deleting"}
config:/Accelerator/Top/sim:\/\/sine: null
```

The end result should be that the alarm server no longer tracks that PV, i.e., it is no longer listed `ls -all` on the alarm server console, and the item is also no longer shown in the alarm tree GUI.
I hope you can confirm that everything "works" from an end-user perspective.

But I agree that you are correct in that the "state:..." message remains in the Kafka topic, so when a client starts up, it will see the last `state:..` update, maybe followed by the "delete" message if that has not, yet, been compacted, and finally the  "null" tombstone. As a result, a client might briefly create and then right away destroy the item.
This could be optimized by deleting like this:

```
# Deletion marker and "tombstone" for both state and config
config:/Accelerator/Top/sim:\/\/sine: {"user":"...","host":"...","delete":"Deleting"}
state:/Accelerator/Top/sim:\/\/sine: null
config:/Accelerator/Top/sim:\/\/sine: null
```

I'll look at adding that.

sasaki77, 2021-06-08:
Thank you for fixing the issue quickly.

closed 2021-06-08 by sasaki77

kasemir, 2021-06-08:
Thanks for your good description of the issue, which made it easy to fix.

For the record, note that the exact order of messages is now like this:
```
# Client which wants to delete an item sends "delete" info and then "null"
config:/Accelerator/Top/sim:\/\/sine: {"user":"...","host":"...","delete":"Deleting"}
config:/Accelerator/Top/sim:\/\/sine: null

# Server sends state update with "null"
state:/Accelerator/Top/sim:\/\/sine: null
```

I figured it's best to stick with the general idea that clients send "config" messages and the alarm server sends "state" updates. So this depends on the alarm server to be running, and the "state: .. null" will be last.
If the alarm server is not running, there will be no "state: .. null" message until the alarm server is started.
Not a problem, just means the situation will be as before until the alarm server is started and then sends the "state: .. null".

Also, if one updates the alarm server but still has older clients running, they will log a warning when they see the "state: .. null" message, but again everything "works" and the warning goes away as soon as you update the client to the latest version of the code.

sasaki77, 2021-06-09:
It was really helpful that you described it in detail.

Let me comment about this.

> If the alarm server is not running, there will be no "state: .. null" message until the alarm server is started.
> Not a problem, just means the situation will be as before until the alarm server is started and then sends the "state: .. null".

Please assume that we ran the alarm server and created `/Accelerator/TEST/Alarm00000` item.
And then we stopped the alarm server and deleted `/Accelerator/TEST/Alarm00000` with `-import` option command.
Kafka messages would be as following.

```json
config:/Accelerator/TEST/Alarm00000: {"user":"sasaki","host":"sasaki-VirtualBox","description":"Alarm00000"}
state:/Accelerator/TEST/Alarm00000: {"severity":"OK","message":"OK","value":"0.0","time":{"seconds":1623204507,"nano":304174601},"current_severity":"OK","current_message":"NO_ALARM"}
config:/Accelerator/TEST/Alarm00000: {"user":"sasaki","host":"sasaki-VirtualBox","delete":"Deleting"}
config:/Accelerator/TEST/Alarm00000: null
```

If there are other config topic messages, Kafka might perfome log comaction.
In this case, we only have "tombstone" message for `config:/Accelerator/TEST/Alarm00000`.

```json
state:/Accelerator/TEST/Alarm00000: {"severity":"OK","message":"OK","value":"0.0","time":{"seconds":1623204507,"nano":304174601},"current_severity":"OK","current_message":"NO_ALARM"}
config:/Accelerator/TEST/Alarm00000: null
```

Alarm server can't send "state: .. null" even if we start it because there's no valid config for `/Accelerator/TEST/Alarm00000`. `state:/Accelerator/TEST/Alarm00000:` will remain forever.

sasaki77, 2021-06-09:
Another case, assume that we created `/Accelerator/TEST/FOO/Alarm00000` and deleted `/Accelerator/TEST/FOO/Alarm00000` and `/Accelerator/TEST/FOO`.
`state:/Accelerator/TEST/FOO/Alarm00000` was deleted with "tombstone", but `state:/Accelerator/TEST/FOO` will remain forever because it's not a PV.

```json
config:/Accelerator/TEST: {"user":"sasaki","host":"sasaki-VirtualBox"}
config:/Accelerator/TEST/FOO: {"user":"sasaki","host":"sasaki-VirtualBox"}
config:/Accelerator/TEST/FOO/Alarm00000: {"user":"sasaki","host":"sasaki-VirtualBox","description":"Alarm00000"}
state:/Accelerator/TEST/FOO: {"severity":"OK"}
state:/Accelerator/TEST: {"severity":"OK"}
state:/Accelerator/TEST/FOO/Alarm00000: {"severity":"OK","message":"OK","value":"0.0","time":{"seconds":1623204989,"nano":268343365},"current_severity":"OK","current_message":"NO_ALARM"}
state:/Accelerator: {"severity":"OK"}
config:/Accelerator/TEST/FOO/Alarm00000: {"user":"sasaki","host":"sasaki-VirtualBox","delete":"Deleting"}
config:/Accelerator/TEST/FOO/Alarm00000: null
config:/Accelerator/TEST/FOO: {"user":"sasaki","host":"sasaki-VirtualBox","delete":"Deleting"}
config:/Accelerator/TEST/FOO: null
state:/Accelerator/TEST/FOO/Alarm00000: null
```

sasaki77, 2021-06-09:
I don't think they're major problems for now, but I left the comments just in case,

kasemir, 2021-06-09:
You're right, in both examples, there is no complete removal of the state: .. topics.
Everything will still "work", but it's a little inefficient because the state: messages will create the alarm tree elements, only to soon be deleted.
One workaround is a periodic cleanup: Once a year, for example during a major outage, export the alarm config, stop alarm server and clients, delete the alarm topics, re-create them, import alarm config, start alarm server and clients.
This doesn't take that long, you can do most via a remote login to the server host. The main issue might be stopping and re-starting the alarm client. It's usually running on one dedicated host in the main control room, so it's just one instance, but requires being in the control room at that host.