ControlSystemStudio/phoebus #1926: Detached display width and height

fredrik-soderberg, 2021-06-10:
When I detach a display in Phoebus, the width and height is calculated from the parent. The width and height shown in the display property panel is not used. Some of our users would prefer to use the values in the property panel. They report that they have to resize a lot of windows all the time when using the detach functionality. What is the main reason for using the current implementation to the determine the width and height? Would it be possible to use the width and height from the display property instead or will it have other consequences/side effects?

kasemir, 2021-06-11:
> Would it be possible ..

Not easily. The 'docking' mechanism has no idea about the content. It doesn't care if the pane contains a display builder runtime or an editor or probe or ...
At the same time, the apps inside the pane have no idea if the pane is one of many tabs, or in a vertically divided tab, or a standalone window.

It's of course possible to change it, but we should NOT change it by making the docking pane check for a display builder runtime as a child, since that would introduce unnecessary coupling. What if you wanted to build a phoebus with only data browser? Couldn't, would now always need to include the display runtime...
So one way to solve this would be to add something like a "Dimensions getSizeHint()" to the app, which the docking lib would ignore most of the time _except_ when it's detaching a pane. Personally, not sure if it's worth it because you might be able to find as many users who would vote for "use the origin window size" option that's currently implemented.

cjenkscybercom, 2022-09-15:
@kasemir Started looking into this one again, I see a year later or so (CSSTUDIO-1121) and can see where you're coming from with this implementation-wise. 

I'm curious if you've had any additional thoughts on this, since I'm working on (potentially) implementing it.

cjenkscybercom, 2022-09-15:
@kasemir It so far wasn't a lot to change, and the ramifications are opt-in...So I made a PR. Would you mind please taking a closer look at it? Thanks!

https://github.com/ControlSystemStudio/phoebus/pull/2387

closed 2022-09-19 by cjenkscybercom