ControlSystemStudio/phoebus #1928: Acknowledging disconnected alarms also acknowledges "proper" alarm state

krisztianloki, 2021-06-10:
If an alarm PV is configured as `latching` and the IOC goes offline the alarm server will show the PV as disconnected even after the IOC comes back online and the PV is in a "proper" alarm. Now, if I acknowledge the alarm to get rid of the disconnected state it will also acknowledge the "proper" alarm that was not even visible before.
I think there are at least two problems with this behavior:

- the alarm server will show latching PVs as disconnected even though the PV is now connected and in an alarm state
- there is no way to determine that the disconnected state is still valid or just latched

shroffk, 2021-06-10:
Hi, 
In general a disconnect is the more sever event and hence it has the highest priority, it makes sense that alarms to should latch on those... if a machine power cycled and now it is in minor alarm state. It might be very important to first check on why the machine went down in the first place.

When acknowledging an alarm you can always check the current state, there is a seperate column in the alarm table
[image]

If the PV is still in an alarm state it will also be visible in the "acknowledged" alarm table view.

The assumption is that if you are acknowledging an alarm then you are "on top of it"

kasemir, 2021-06-11:
What you describe pretty much exactly matches the example in testLatchedAnnunciatedAlarmUndefined()  https://github.com/ControlSystemStudio/phoebus/blob/dbe35f553395e8d1182266b0d5116dc088fc0717/services/alarm-server/src/test/java/org/phoebus/applications/alarm/server/AlarmLogicUnitTest.java#L266.
It behaves exactly as designed, and has been doing this from the start of the alarm system, back when it was Eclipse-based. Kunal gives a good explanation as to why it was designed this way.

krisztianloki, 2021-06-11:
I was looking at the `Alarm tree` and it was not immediately clear to me that the current alarm severity is displayed at the end of the line.
Still, I think that it should be possible to acknowledge PV disconnection without acknowledging any current alarm. For example after a routine maintenance window every alarm will be disconnected and it is easier to just acknowledge the root component instead of acknowledging every single one individually.
Also, instead of a single disconnected PV "hijacking" the component alarm states up in the tree overriding any "actual" alarms what about using something like the "acknowledged but still alive" alarm badge for the "there are disconnected alarms and there are major/minor alarms in the tree" case?

shroffk, 2021-06-11:
The situation you described perfectly fits the use of the maintenance mode
[image]

In the maintenance mode "proper" alarms will not be ack'ed and the behaviour of the alarm system will be exactly as you requested

krisztianloki, 2021-06-11:
Did not know about this feature. Thanks!

closed 2021-06-11 by krisztianloki