ControlSystemStudio/phoebus #1930: Customization

lorenzo-gomez-windhover, 2021-06-10:
How would one go about changing the theme to say something  like "dark mode" as opposed to the white theme at Phoebus has at the moment?

Is this supported at this point?

Are there plans to do so?

Thanks!
Lorenzo

georgweiss, 2021-06-15:
I'd say there are two sides to this:
1) Styling of OPIs, which is controlled by the user. Most styling aspects related to color can be defined by the user, but there is no mechanism to define for instance two different color schemes between which the user may toggle.
2) Styling of non-OPI elements like the application title bar, toolbar and toolbar button styles, tab header style etc. These elements can be customized using CSS (Cascading Style Sheets) as supported by JavaFX, see for instance https://stackoverflow.com/questions/49159286/make-a-dark-mode-with-javafx. But Phoebus has - to my knowledge - no API to switch to a different set of CSS files in order to switch color scheme at runtime.

lorenzo-gomez-windhover, 2021-06-15:
I will look into these solutions.
Thank you!!!