ControlSystemStudio/phoebus #1932: Is there a concept of a "project"?

lorenzo-gomez-windhover, 2021-06-15:
Hi there,

I've been playing around with phoebus and it has been a very pleasant user-experience. One thing that I wonder about that existed in the old CS Studio was the concept of "projects". I know this was just an artifact of it being Eclipse-based, and I was trying to look for ways to look for a project by using the tool and even reading the documentation, but couldn't find anything. The only way I have found to load a "project" is to to use the File Browser tool and load the root display of the folder directly from the file system. 

Is there a better way of loading a "project", similar to the Eclipse way of doing it?

Hope this questions makes sense.

Thanks!
Lorenzo

georgweiss, 2021-06-15:
A number of the concepts supported natively in Eclipse ("project" being one of them) have no counterpart in Phoebus. The File Browser is indeed your access to the display files.
Note that a display can be "launched" using an application argument identifying the wanted display, see class [Launcher](https://github.com/ControlSystemStudio/phoebus/blob/master/core/launcher/src/main/java/org/phoebus/product/Launcher.java)

kasemir, 2021-06-16:
Right, there is no "workspace" or "project".
For some users, the "workspace" was confusing because what you saw there was different from the actual file system as seen by all the other tools. So now phoebus simply uses the file system.
Still, for some other users it was helpful to have the "workspace" as a starting point, seeing only files in there that they were meant to see. To mimic that, you can simply create some top-level workspace directory in the file system, populate that with display files or sub-folders or maybe soft-links to other files/folders, and then configure the file browser in phoebus to start in that top-level directory. So users who liked the workspace can start the file browser and it looks similar, except that the path on top tells you exactly where you are in the real file system, and you're free to navigate anywhere.

See https://control-system-studio.readthedocs.io/en/latest/preference_properties.html#filebrowser