ControlSystemStudio/phoebus #1934: Alarm logger: Pushing messages into Elasticsearch is a little bit slow

sasaki77, 2021-06-17:
I'm testing alarm server and alarm logger with 50,000 PVs. All PVs are running on one IOC.
Once the IOC is stopped, 50,000 state change messages are issued.
Alarm logger receives the messages and pushes them into Elasticsearch, but it needs a lot of time to complete pushing all messages.

I configured Elasticsearch with Ingest pipelines to measure a time to complete it as following. We can see the time when the document is stored with this.

```bash
curl -X PUT "http://localhost:9200/_ingest/pipeline/indexed_at?pretty" -H 'Content-Type: application/json' -d '
{
  "description": "Adds indexed_at timestamp to documents",
  "processors": [
    {
      "set": {
        "field": "_source.indexed_at",
        "value": "{{_ingest.timestamp}}"
      }
    }
  ]
}
'
```

Following is the last document stored into Elasticsearch when the 50,000 state change messages are issued. `indexed_at` field indicates that it takes about 7 minutes to store the document.

```bash
$ curl -X GET 'http://localhost:9200/accelerator_alarms_state*/_search?q=*&format=json&pretty&sort=indexed_at:desc&from=0&size=1'
...
{
  "_index" : "accelerator_alarms_state_2021-05-31",
  "_type" : "alarm",
  "_id" : "siVKCXoBi0cjCWSAtncP",
  "_score" : null,
  "_source" : {
    "severity" : "UNDEFINED",
    "pv" : "Alarm48096",
    "message" : "Disconnected",
    "notify" : "true",
    "current_message" : "Disconnected",
    "mode" : null,
    "message_time" : "2021-06-14 06:44:50.660",
    "current_severity" : "UNDEFINED",
    "indexed_at" : "2021-06-14T06:52:09.103327Z",
    "latch" : "false",
    "time" : "2021-06-14 06:44:50.659",
    "config" : "state:/Accelerator/TEST/Alarm48096",
    "value" : ""
  },
  "sort" : [
    1623653529103
  ]
}
...
```

I don't think this is caused by settings of Elasticsearch because I confirmed Logstash can smoothly push all 50,000 messages into Elasticsearch.

Current implementation creates IndexRequest for Elasticsearch per message, right?
I guess we can improve alarm logger with using Elasticsearch Bulk API, but I have no idea to use Bulk API with Kafka stream API.

I agree we should avoid alarm flood and this test case is really rare situation, but it would be helpful if alarm logger could process messages smoothly.

kasemir, 2021-06-17:
> Current implementation creates IndexRequest for Elasticsearch per message, right?

I think you're correct. If I understand this correctly, each message eventually ends in `indexAlarmStateDocument` https://github.com/ControlSystemStudio/phoebus/blob/3e468cf6b913e2a21ffabbce7b0443147386b1cb/services/alarm-logger/src/main/java/org/phoebus/alarm/logging/ElasticClientHelper.java#L94.
Since the basic interface to elastic is a web service, pushing a single message per web request is likely the bottleneck in your test scenario.  @shroffk would know more about how one might collect a few messages for a potential bulk insert. I assume it would require added logic to collect messages, submitting them whenever N messages have accumulated or some time elapsed.

> I confirmed Logstash can smoothly push all 50,000 messages into Elasticsearch

Was that an unrelated test of pushing this amount of demo messages into ES, or did you somehow get data for the actual alarm messages into Logstash and then on into ES?

sasaki77, 2021-06-17:
The Logstash test was done with actual alarm messages.

I prepared following Logstash pipeline. The stored documents with this pipeline is not exactly same as Alarm logger, so the documents were pushed into another index. I didn't prepare index template for this. The documents might be stored with dynamic mapping.

```
input {
    kafka {
        bootstrap_servers => "localhost:9092"
        topics => ["Accelerator"]
        auto_offset_reset => "latest"
        decorate_events => "basic"
    }
}

filter {
  json{
    source => "message"
  }

  # Drop root state message
  if ![time] {
    drop {}
  }

  mutate {
    add_field => {
      "message_time" => "%{[@metadata][kafka][timestamp]}"
      "config" => "%{[@metadata][kafka][key]}"
      "pv" => "%{[@metadata][kafka][key]}"
    }
  }

  mutate {
    gsub => [
      "pv","^.*/",""
    ]
  }

  if ![notify] {
    mutate {
      add_field => {
        "notify" => "true"
      }
    }
  }

  if ![mode] {
    ruby { code => 'event.set("mode", nil)' }
  }

  if ![latch] {
    mutate {
      add_field => {
        "latch" => "false"
      }
    }
  }

  ruby { code => 'event.set("[time][nano]", format("%09<number>d", number: event.get("[time][nano]")))' }
  ruby { code => 'event.set("[time][nano]", event.get("[time][nano]").slice!(0,3))' }

  mutate {
    replace => {
      "time" => "%{[time][seconds]}%{[time][nano]}"
    }
  }

  date {
    match => [ "message_time", "UNIX_MS" ]
    target => "message_time"
  }

  date {
    match => [ "[time]", "UNIX_MS" ]
    target => "time"
  }

}

output {
   elasticsearch {
     hosts => ["localhost:9200"]
     index => "logstash-state-%{+YYYY.MM}"
   }
}
```

Following is the last message from 50,000 PVs disconnection event. It indicates Logstash processed the last message in 100 ms.

```bash
$ curl -X GET 'http://localhost:9200/logstash-state*/_search?q=*&format=json&pretty&sort=indexed_at:desc&from=0&size=1'
...
{
  "_index" : "logstash-state-2021.06",
  "_type" : "doc",
  "_id" : "m7AHE3oBMoch0w1N6ZYP",
  "_score" : null,
  "_source" : {
    "severity" : "UNDEFINED",
    "pv" : "Alarm37213",
    "message" : "Disconnected",
    "notify" : "true",
    "current_message" : "Disconnected",
    "mode" : null,
    "message_time" : "2021-06-16T04:15:23.282Z",
    "current_severity" : "UNDEFINED",
    "@timestamp" : "2021-06-16T04:15:23.301Z",
    "indexed_at" : "2021-06-16T04:15:23.407091Z",
    "@version" : "1",
    "latch" : "false",
    "time" : "2021-06-16T04:15:23.282Z",
    "value" : "",
    "config" : "state:/Accelerator/TEST/Alarm37213"
  },
  "sort" : [
    1623816923407
  ]
}
...
```

it's difficult to simply compare Alarm logger and Logstash result, but I think Alarm logger can be improved.

shroffk, 2021-06-17:
Hello,

Bulk Inserts would be the best solution to address the scenario.
Here is an example of how I use it in ChannelFinder where there is a use case where hundred/thousands of channels need to be indexed at once.
https://github.com/ChannelFinder/ChannelFinderService/blob/ef86257756e1c594bc23186ea90e4375f426b0c9/src/main/java/gov/bnl/channelfinder/ChannelRepository.java#L120

It would be trivial to add a bulk insert to the AlarmMessageProcessor... but the more complicated bit would be to add aggregation and caching in kafka streams. Currently, the architecture is quite simple...we have a processor registered which processes each message. This covers the most common use cases but I agree this can be updated.

When I was writing this I did not find any much support for batch processing
https://issues.apache.org/jira/browse/KAFKA-7432
but there might be more solutions now...

shroffk, 2021-06-17:
I think I might have a simple solution... let me prepare a patch

shroffk, 2021-06-17:
So, I have added a simple queue which collects events into 250ms buckets and creates a single bulk request to insert them.
I also removed the post creation validation cause it really isn't very useful.. if something going wrong the exception logging should be enough.

I have pushed it to a separate branch so you could test it.

But in addition to the elastic logging, the processing of the alarm messages also has some cost. 

In your above example, do you have messages with config
"config" : "state:/Accelerator/TEST/Alarm37213"

shroffk, 2021-06-17:
@kasemir I was wondering if it would be ok to disallow "," in the alarm pv names.

my usecase is to support the new formula functions 
```
=majorAlarm('ca://some_pv'==match_some_condition,"A alarmpv which hasn't made it to an IOC")
```

In this formula there are 2 arguments which are seperated with a ","

sasaki77, 2021-06-18:
> In your above example, do you have messages with configs as follows?
> "config" : "state:/Accelerator/TEST/Alarm37213"
> "config" : "state:/Accelerator/TEST"
> "config" : "state:/Accelerator"

I'm not sure if I understand your question. Above Logstash example drops messages without time field, so messages like below are dropped.

`state:/Accelerator/TEST: {"severity":"UNDEFINED"}`
`state:/Accelerator: {"severity":"UNDEFINED"}`

kasemir, 2021-06-18:
> if it would be ok to disallow "," in the alarm pv names.

Not sure.
"Real PVs" as used with Channel Access won't contain ",", so for the most part, you'd be OK.

Simulated PVs can: `sim://sine(-5, 5, 0.5)`, or `loc://array(1, 2, 3, 4)`. Might be OK because simulated/local/.. PVs aren't useful for an operational alarm setup.

Not sure, however, what the relatively new server-side IOC filters use for a syntax.

sasaki77, 2021-06-18:
I have tested alarm logger with your branch. I produced 50,000 PVs disconnection.

Kafka consumer received following messages.

```json
...
CreateTime:1623978767891: state:/Accelerator/TEST/Alarm18435: {"severity":"UNDEFINED","message":"Disconnected","value":"","time":{"seconds":1623978767,"nano":891475000},"current_severity":"UNDEFINED","current_message":"Disconnected"}
CreateTime:1623978767894: state:/Accelerator/TEST/Alarm16135: {"severity":"UNDEFINED","message":"Disconnected","value":"","time":{"seconds":1623978767,"nano":894028000},"current_severity":"UNDEFINED","current_message":"Disconnected"}
CreateTime:1623978777933: state:/Accelerator: {"severity":"UNDEFINED"}
CreateTime:1623978787968: state:/Accelerator: {"severity":"UNDEFINED"}
CreateTime:1623978798024: state:/Accelerator: {"severity":"UNDEFINED"}
...
```

The last Kafka message `Alarm16135` was indexed really quickly.

```bash
$ curl -X GET 'http://localhost:9200/accelerator_alarms_state*/_search?q=*&format=json&pretty&sort=message_time:desc&from=0&size=1'
...
{
  "_index" : "accelerator_alarms_state_2021-05-31",
  "_type" : "alarm",
  "_id" : "JPmtHHoB-TNhlrMjdo2I",
  "_score" : null,
  "_source" : {
    "severity" : "UNDEFINED",
    "pv" : "Alarm16135",
    "message" : "Disconnected",
    "notify" : "true",
    "current_message" : "Disconnected",
    "mode" : null,
    "message_time" : "2021-06-18 01:12:47.894",
    "current_severity" : "UNDEFINED",
    "indexed_at" : "2021-06-18T01:12:48.008403Z",
    "latch" : "false",
    "time" : "2021-06-18 01:12:47.894",
    "config" : "state:/Accelerator/TEST/Alarm16135",
    "value" : ""
  },
...
```

shroffk, 2021-06-18:
Great,
Thanks for noticing and helping test this :)
I will open a PR

closed 2021-06-25 by sasaki77