ControlSystemStudio/phoebus #1936: Alarm: CA connection becomes unstable when PVs are deleted

sasaki77, 2021-06-21:
**Bug description**
CA connection between Alarm server and IOC becomes unstable when Alarm PVs are deleted.
The CA connection becomes disconnected and connected repeatedly.

**To Reproduce**
1. There are 2 IOCs. IOC A has 500 PVs, IOC B has 49500 PVs.
2. Alarm server is configured to monitor 50000 PVs (500 + 49500) in 2 IOCs.
3. Start Alarm server.
4. Open Phoebus Alarm tree GUI. Delete 400 Alarm PVs running on IOC A.
5. The CA connection becomes unstable. The CA connection between Alarm server and IOC A becomes disconnected and connected repeatedly.
6. The CA connection between Alarm server and IOC B has no problem.

**Workaround**
Comment out these lines not to send config delete message.

https://github.com/ControlSystemStudio/phoebus/blob/3d46002b4c58d06aeec97d0eecdf84d5e7332cf3/app/alarm/model/src/main/java/org/phoebus/applications/alarm/client/AlarmClient.java#L542-L544

Alarm server stops and starts alarm PV when it receives Config delete message. Then It stops alarm PV when it receives the tombstone marker.
Following is an example log.

```
2021-06-21 11:13:37 FINE [org.phoebus.applications.alarm] Stop Alarm00000
2021-06-21 11:13:37 FINE [org.phoebus.applications.alarm] Start Alarm00000
2021-06-21 11:13:37 FINE [org.phoebus.applications.alarm] Detach /Accelerator/TEST/Alarm00000 from parent
2021-06-21 11:13:37 FINE [org.phoebus.applications.alarm] Stop Alarm00000
```

We can just stop the alarm PVs if config delete messages is not sent.
I'm not sure why, but we can delete Alarm PVs without problem in this situation.

**Additional context**
- This problem tend to happen when Alarm server monitors tens of thousands PVs even if IOC has only hundreds PVs that contain the PVs to be removed.
  - For example, assume Alarm server is configured to monitor only 500 PVs running on IOC A. We can delete 400 Alarm PVs without ploblems in this situation.
- Unstable connection doesn't affect any other connections.
  - Alarm server can continue to connect IOC B without problems in **To Reproduce** example.

**Logs when PVs are deleted**
Alarm server log
```
2021-06-21 11:13:37 FINE [org.phoebus.applications.alarm] Stop Alarm00000
2021-06-21 11:13:37 FINE [org.phoebus.applications.alarm] Start Alarm00000
2021-06-21 11:13:37 FINE [org.phoebus.applications.alarm] Detach /Accelerator/TEST/Alarm00000 from parent
2021-06-21 11:13:37 FINE [org.phoebus.applications.alarm] Stop Alarm00000
2021-06-21 11:13:37 FINE [org.phoebus.applications.alarm] Stop Alarm00001
2021-06-21 11:13:37 FINE [org.phoebus.applications.alarm] Start Alarm00001
2021-06-21 11:13:37 FINE [org.phoebus.applications.alarm] Detach /Accelerator/TEST/Alarm00001 from parent
2021-06-21 11:13:38 FINE [org.phoebus.applications.alarm] /Accelerator/TEST/Alarm00491 disconnected
2021-06-21 11:13:38 FINER [org.phoebus.applications.alarm] /Accelerator/TEST/Alarm00491 changes to UNDEFINED/Disconnected (), 2021-06-21 11:13:38.024, UNDEFINED/Disconnected (), 2021-06-21 11:13:38.024
2021-06-21 11:13:38 FINE [org.phoebus.applications.alarm] /Accelerator/TEST/Alarm00459 disconnected
2021-06-21 11:13:38 FINER [org.phoebus.applications.alarm] /Accelerator/TEST/Alarm00459 changes to UNDEFINED/Disconnected (), 2021-06-21 11:13:38.029, UNDEFINED/Disconnected (), 2021-06-21 11:13:38.029
2021-06-21 11:13:38 WARNING [org.phoebus.pv] Channel Access Exception from com.cosylab.epics.caj.CAJContext@67f275e0 for channel CAJChannel = { name = Alarm00001, connectionState = CONNECTED }: User destroyed channel
2021-06-21 11:13:38 FINE [org.phoebus.applications.alarm] Stop Alarm00001
2021-06-21 11:13:38 FINE [org.phoebus.applications.alarm] /Accelerator/TEST/Alarm00435 disconnected
2021-06-21 11:13:38 FINER [org.phoebus.applications.alarm] /Accelerator/TEST/Alarm00435 changes to UNDEFINED/Disconnected (), 2021-06-21 11:13:38.063, UNDEFINED/Disconnected (), 2021-06-21 11:13:38.063
2021-06-21 11:13:38 FINER [org.phoebus.applications.alarm] /Accelerator/TEST/Alarm00491 changes to OK/NO_ALARM (0.0), 2021-06-21 11:13:37.926, OK/NO_ALARM (0.0), 2021-06-21 11:13:37.926
2021-06-21 11:13:38 FINE [org.phoebus.applications.alarm] /Accelerator/TEST/Alarm00290 disconnected
2021-06-21 11:13:38 FINER [org.phoebus.applications.alarm] /Accelerator/TEST/Alarm00290 changes to UNDEFINED/Disconnected (), 2021-06-21 11:13:38.099, UNDEFINED/Disconnected (), 2021-06-21 11:13:38.099
2021-06-21 11:13:38 FINER [org.phoebus.applications.alarm] /Accelerator/TEST/Alarm00491 received VDouble[0.0, NONE(NONE) - NO_ALARM, 2021-06-21T02:13:37.926066615Z] -> Current: OK/NO_ALARM (0.0), 2021-06-21 11:13:37.926, Alarm: OK/NO_ALARM (0.0), 2021-06-21 11:13:37.926
2021-06-21 11:13:38 FINE [org.phoebus.applications.alarm] /Accelerator/TEST/Alarm00386 disconnected
2021-06-21 11:13:38 FINER [org.phoebus.applications.alarm] /Accelerator/TEST/Alarm00386 changes to UNDEFINED/Disconnected (), 2021-06-21 11:13:38.129, UNDEFINED/Disconnected (), 2021-06-21 11:13:38.129
2021-06-21 11:13:38 FINE [org.phoebus.applications.alarm] /Accelerator/TEST/Alarm00005 disconnected
2021-06-21 11:13:38 FINER [org.phoebus.applications.alarm] /Accelerator/TEST/Alarm00005 changes to UNDEFINED/Disconnected (), 2021-06-21 11:13:38.141, UNDEFINED/Disconnected (), 2021-06-21 11:13:38.141
2021-06-21 11:13:38 SEVERE [com.cosylab.epics.caj.CAJChannel]
java.lang.IllegalStateException: transport closed
        at com.cosylab.epics.caj.impl.CATransport.submit(CATransport.java:908)
        at com.cosylab.epics.caj.impl.requests.AbstractCARequest.submit(AbstractCARequest.java:88)
        at com.cosylab.epics.caj.CAJMonitor.clear(CAJMonitor.java:141)
        at com.cosylab.epics.caj.CAJChannel.destroyAllMonitors(CAJChannel.java:1213)
        at com.cosylab.epics.caj.CAJChannel.destroyChannel(CAJChannel.java:414)
        at com.cosylab.epics.caj.CAJContext.destroyChannel(CAJContext.java:1069)
[... 134 lines cut ...]
CAS: forcing disconnect from 127.0.0.1:38508
CAS: request from 127.0.0.1:38510 => bad resource ID
CAS: Request from 127.0.0.1:38510 => cmmd=15 cid=0xb7d type=34 count=1 postsize=0
CAS: Request from 127.0.0.1:38510 =>   available=0x18c7e 	N=0 paddr=(nil)
CAS: forcing disconnect from 127.0.0.1:38510
CAS: request from 127.0.0.1:38512 => bad resource ID
CAS: Request from 127.0.0.1:38512 => cmmd=2 cid=0x569 type=20 count=1 postsize=0
CAS: Request from 127.0.0.1:38512 =>   available=0x6a7 	N=0 paddr=(nil)
CAS: forcing disconnect from 127.0.0.1:38512
CAS: request from 127.0.0.1:38514 => bad resource ID
CAS: Request from 127.0.0.1:38514 => cmmd=15 cid=0xce0 type=34 count=1 postsize=0
CAS: Request from 127.0.0.1:38514 =>   available=0x18ca9 	N=0 paddr=(nil)
CAS: forcing disconnect from 127.0.0.1:38514
CAS: forcing disconnect from 127.0.0.1:38516
CAS: request from 127.0.0.1:38518 => bad resource ID
CAS: Request from 127.0.0.1:38518 => cmmd=2 cid=0xe4a type=20 count=1 postsize=0
CAS: Request from 127.0.0.1:38518 =>   available=0x6b9 	N=0 paddr=(nil)
CAS: forcing disconnect from 127.0.0.1:38518
CAS: request from 127.0.0.1:38520 => bad resource ID
CAS: Request from 127.0.0.1:38520 => cmmd=1 cid=0xf2b type=20 count=1 postsize=16
CAS: Request from 127.0.0.1:38520 =>   available=0x818 	N=0 paddr=(nil)
CAS: forcing disconnect from 127.0.0.1:38520
CAS: forcing disconnect from 127.0.0.1:38522
...
```

kasemir, 2021-06-24:
> Alarm server stops and starts alarm PV when it receives Config delete message.
> Then It stops alarm PV when it receives the tombstone marker.

The delete message is meant to be informational, allowing a logger to track what user, on what host, deleted a channel. The actual deletion is the `null` tombstone.

If the alarm server reacts to both, that would be a bug.
The alarm server needs to react to config changes (stop PV, apply config changes, then restart the PV), but maybe it's considering the 'delete' message a config change when it should better ignore the delete message, and then just delete the PV when the tombstone is received.

kasemir, 2021-07-01:
Updated alarm server so it will ignore the informational config "delete" message, avoiding that extraneous stop/restart.
It will only react to the final config `null` message, deleting the PV once.
That should avoid the issue you see, just as removing the "delete" message did, yet keeping the "delete" message for informational purpose (to log who triggered the deletion).

It doesn't fully explain what's happening on the channel access level. Unclear what causes the "bad resource ID" which then results in the IOC closing the network connection, leaving all channels from the IOC being disconnected until the client tries to re-connect. That would best be explored with a simpler test that just tries to stop/start many PVs without all the remaining alarm logic. Can you provide the script you use to create those many Alarm00000 records?

sasaki77, 2021-07-02:
I confirmed that the alarm server can delete alarm PVs stably.

The script is available from the following link. The db file for 50000 alarm PVs is in `demo` directory.

https://github.com/sasaki77/alarm-test-script

sasaki77, 2021-07-08:
I'm closing this issue for now.  Connection problem still remain, but that might be a connection layer problem not the alarm server problem.

closed 2021-07-08 by sasaki77

kasemir, 2021-07-08:
I agree that the alarm server is now avoiding the issue, but it might still exist.
I plan to look at that within the upcoming weeks with test code that connects/disconnects in a manner similar to what the alarm server did before we improved the 'delete' handling.

kasemir, 2021-07-09:
So far haven't been able to reproduce the "bad resource ID" on the IOC which caused it to close the connection, leading to the follow-up problems.
What version of EPICS base are you using to run the IOC?

sasaki77, 2021-07-12:
I have tested with EPICS R3.15.7. The test environment is as followings.

- OS: Ubuntu 20.04.2 LTS (running on VirtualBox)
- Java: openjdk version "14.0.2"