ControlSystemStudio/phoebus #1941: Question: Alarm Logger with ES 7?

tanviash, 2021-06-28:
Has anyone tried running the alarm-logger service with ES7?

kasemir, 2021-06-28:
https://github.com/ControlSystemStudio/phoebus/blob/master/services/alarm-logger/README.md mentions that it won't work with 7, needs 6. I think there was a ticket with details, but can't find that any more...

shroffk, 2021-06-28:
Hello, 
Alarm logger will not run with ES7 out of the box.
There are 2 sets of changes...the mapping needs to be updated to not use type and secondly the query code will also need to updated to remove the type reference.

In the alarm logger since we are using 1 type per index we are already doing the correct thing according to lucene... however we will still need to make the above minor changes. Instead of doing this for ES7 and then again for ES8 I recommend we leapfrog ES7 and directly move to ES8.

kasemir, 2021-06-28:
Now remember the related issue ticket, which was for the Channel Finder, https://github.com/ChannelFinder/ChannelFinderService/issues/20. It has similar ES7 issues, and for the CF there are workarounds: Adding `?include_type_name=true` to URLs etc. Not sure if it makes sense to try that for the alarm logger as well.

tanviash, 2021-06-28:
I tried the ?include_type_name=true option with the 3 URLs for ES template creation in https://github.com/ControlSystemStudio/phoebus/blob/master/services/alarm-logger/startup/create_alarm_template.sh

and was able to only read back command messages that come from alarms_cmd index (See attached pic). Nothing showed up in logger client for state and config messages even though all 3 types of indices got created and has correct data inside them. 

[image]

tanviash, 2021-07-12:
Closing this issue. After a fresh startup of alarm logger service and ES 7, include_type_name=true option works fine with alarm logger and client to read and write to ES7.

closed 2021-07-12 by tanviash