ControlSystemStudio/phoebus #1943: Broken link in documentation

thomasms, 2021-07-01:
When trying to view the raw rst file for the documentation, I tried the link circled in red below, but the link seems broken. It is not clear to me where the correct file lies on the repository.

[image]

kasemir, 2021-07-01:
Unfortunately, read-the-docs assumes that all documentation is in one large GitHub repo just for the *.rst files, but we assemble the doc from app-specific sections.
Documentation is in `doc` folders next to the corresponding tool, in this case https://github.com/ControlSystemStudio/phoebus/tree/master/services/alarm-config-logger/doc

For more see https://github.com/ControlSystemStudio/phoebus/tree/master/docs

thomasms, 2021-07-01:
Fair enough. I made a pull request to fix the small but annoying grammatical error in the documentation, if this helps.
#1944

closed 2021-07-01 by kasemir

thomasms, 2021-07-02:
Happy to help.