ControlSystemStudio/phoebus #1946: OPI auto conversion support for Phoebus Actions

shroffk, 2021-07-02:
In order to support the migration from BOY to Display Builder, I had added a new action "OPEN_PHOEBUS" which allows the creation of action and menu button actions which open the specified display file in Phoebus.

When these screens are converted to BOB or opened in Phoebus the "OPEN_PHOEBUS" should be replaced by "OPEN_DISPLAY"

shroffk, 2021-07-02:
@kasemir is there a more elegant solution to this?
I thought this was the simplest way.
```

            // Support for translating the cs-studio actions to launch phoebus to simple open display actions
            final Element original_actions = XMLUtil.getChildElement(xml, "actions");
            if(original_actions != null)
            {
                NodeList list = original_actions.getElementsByTagName("action");
                for (int i = 0; i < list.getLength(); i++)
                {
                    //list.item(i).
                    Node node = list.item(i);
                    if(node.getAttributes().getNamedItem("type").getNodeValue().equalsIgnoreCase("OPEN_PHOEBUS"))
                    {
                        node.getAttributes().getNamedItem("type").setNodeValue("OPEN_DISPLAY");
                    }
                }
            }

```
Also, I am not sure why the line endings were not unix...my push changed that but I can undo it

kasemir, 2021-07-02:
This would be a tad shorter because it removes the "int i", but yours looks good to me as well.
```
            final Element original_actions = XMLUtil.getChildElement(xml, "actions");
            if (original_actions != null)
                for (Element action : XMLUtil.getChildElements(original_actions, XMLTags.ACTION))
                {
                    Element type = XMLUtil.getChildElement(action, "type");
                    if (type.getNodeValue().equalsIgnoreCase("OPEN_PHOEBUS"))
                        type.setNodeValue("OPEN_DISPLAY");
                }
```

kasemir, 2021-07-02:
Scratch that, type is an attribute, not a node, right? So use what you have!

closed 2021-07-06 by shroffk