ControlSystemStudio/phoebus #1956: Migration CS-Studio XY plot to Strip chart in Phoebus

geniejhang, 2021-07-21:
Hi,

I recently handling one OPI having XY plot to display time series of a PV.
While I'm doing that, I found some difference between two.
I wonder if they're intended or just missed.

[image]

So, left one is Phoebus and the right is CS-Studio.

1. The numbers shown on Y axis are decimal format even when they're small. There's no option to change the display format to scientific notation.
2. Gray strip at the right end of Phoebus Strip chart appears and keeps changing its width. In my opinion, it's there for no reason.

Last one is the update rate.

https://user-images.githubusercontent.com/5515875/126512579-35a7476d-9001-4cea-80cb-6cb0cf346586.MOV

As you can see from the video above, update rate is slightly different. I guess 2 fps in CS-Studio and 1 fps in Phoebus.
Combined with the gray strip in Phoebus, it shows seemingly low performance. Is it the limitation of JavaFX?

Thanks.

kasemir, 2021-07-21:
You bring up several points that might be better handled separately, but here goes..

When scrolling such that the right edge is always "now", the horizontal axis keeps moving, the labels might jump to new positions, the redraw takes a little longer, and some viewers might get dizzy.
The plot therefore differentiates between scrolling and updating. It might update at 10 Hz, but only move the horizontal axis once every 3 seconds. This means at the right edge there's a section that contains the "future" which will fill for 3 seconds. With the data browser, both the redraw rate and the scroll rate can be configured. If you don't like the defaults, change them.
To configure the color of the gray stripe, see https://control-system-studio.readthedocs.io/en/latest/preference_properties.html#javafx-rtplot:
```
# If you prefer to not highlight the plot region beyond 'now'
# shady_future=128, 128, 128, 0
```

For overall display update throttling, see these settings:
```
# Faster (for tests)
# Display actual update rate by setting
# -Dorg.csstudio.javafx.rtplot.update_counter=true
org.csstudio.display.builder.runtime/update_throttle=1
org.csstudio.display.builder.representation/update_accumulation_time = 1
org.csstudio.display.builder.representation/update_delay = 1
org.csstudio.display.builder.representation/plot_update_delay = 1
org.csstudio.display.builder.representation/image_update_delay = 1
```

The defaults are not optimized for fastest update rate but for lowering CPU usage.

The axes do automatically switch to exponential format as the value range gets large or small enough:

[image]

You could try to come up with better switch-over points.
Yes, one could add many more configuration parameters. But then you would also have to add them do the display builder web runtime, and in about 5 years to the next display tool. I'd prefer to reduce the number of config options and instead keep the display minimal such that they can be easily handled by the next display tool.

geniejhang, 2021-07-21:
Thanks for the answers.
I figured the numbers in throttling setting are in ms. So setting those to 1 means updates in 1kHz.
I tried, but it doesn't look the settings are applied. I put "-Dorg.csstudio.javafx.rtplot.update_counter=true" when running as well.

In the databrowser peroperty panel, I'm able to change "Plot redraw period" and "Scroll Step" to 1 and it worked.
It looks like Scroll Step only takes the number above 1 while I want 0.5. Anyway, this is fine.
Did I do something wrong for making Strip chart the same as databrowser "Plot redraw period = 0.5" and "Scroll step = 1"?

By the way, the Y axis number format is not changing to scientific notation.
The values I'm plotting is about 1E-16 order and still it shows 0.00000000000000004 or so.

Thanks.

kasemir, 2021-07-21:
Yes, the scroll step is internally limited to at least 1 second. You could fix that in https://github.com/ControlSystemStudio/phoebus/blob/2f5de05be5f2827f095097a7963114b17a4efb31/app/databrowser/src/main/java/org/csstudio/trends/databrowser3/model/Model.java#L630

The RtPlot has an internal update throttle that defaults to 200ms = 5 Hz, https://github.com/ControlSystemStudio/phoebus/blob/2f5de05be5f2827f095097a7963114b17a4efb31/app/rtplot/src/main/java/org/csstudio/javafx/rtplot/internal/PlotCanvasBase.java#L202

When you check who's calling `setUpdateThrottle` to change that, you'll see that the XY Plot used by the display builder sets it from  `org.csstudio.display.builder.representation/plot_update_delay = 1`, but the data browser leaves it at the default, which is plenty for the scroll step of 1 second.

You could lower the limit for the data browser scroll step, and maybe update the data browser so it calls the RtPlot `setUpdateThrottle` to for example the scroll step divided by 10?

kasemir, 2021-07-21:
> By the way, the Y axis number format is not changing to scientific notation.
.. for negative exponents ..
You could fix that here, which currently just checks for positive exponents:

https://github.com/ControlSystemStudio/phoebus/blob/2f5de05be5f2827f095097a7963114b17a4efb31/app/rtplot/src/main/java/org/csstudio/javafx/rtplot/internal/LinearTicks.java#L99

You could for example try
```
 final boolean use_exp_notation = order_of_magnitude > exponential_threshold  ||
                                                        order_of_magnitude < -exponential_threshold;
```
or, since the "-" in the exponent takes up space compared to positive exponents, maybe switch earlier?
```
 final boolean use_exp_notation = order_of_magnitude > exponential_threshold  ||
                                                        order_of_magnitude < -exponential_threshold+1;
```

geniejhang, 2021-07-21:
Thanks, @kasemir. If it needs to change something in code level, I would keep it in our fork version.

closed 2021-07-21 by geniejhang