ControlSystemStudio/phoebus #1960: Invalid index : 5 must be within the label range

mdavidsaver, 2021-07-26:
I'm running into an issue with enums.  When connecting a [choice button](https://github.com/ControlSystemStudio/phoebus/files/6879842/as7262ioc.bob.txt) to an [mbbo](https://github.com/ControlSystemStudio/phoebus/files/6874715/enumbug.db.txt) with 4 choices (loaded into softIoc) I sometimes am get an error when attempting to click the 3rd or 4th choices.  eg,

```
Invalid index : 5 must be within the label range [1.0x, 3.7x, 10.0x, 64.0x]
```

Steps to reproduce:

1. With softIoc stopped.  Start phoebus and open the screen, or reload screen if already running
2. Start softIoc.  After reconnect, click on `64.0x` choice.
3. Observe that Text Update above shows `64.0x`
4. Restart softIoc.  After reconnect, client `64.0x` again
6. Observe that Text Update now shows `Invalid index : 5 must be within the label range [1.0x, 3.7x, 10.0x, 64.0x]`
7. Restart softIoc.  After reconnect, client `64.0x` a third time.
8. Observe that Text Update now shows `Invalid index : 7 must be within the label range [1.0x, 3.7x, 10.0x, 64.0x]`.

Each cycle the invalid index being written increases by 2.

If I reload the OPI while the softIoc is running, then things seem to work as expected.  With only valid indices being written.

mdavidsaver, 2021-07-26:
I also notice that while the choice button has `Items from PV` property checked, I see some effect if I adjust the `Items` count.  With the default (2) I see the behavior described above.  If I change `Items` to 3, I see invalid indices, but these increase only by 1 each cycle.  Changing to 4 seems to result invalid indices.

kasemir, 2021-07-26:
The Combo Box works fine, so it seems specific to the choice button, something in there is broken.
I'll look at it.

closed 2021-07-26 by kasemir

mdavidsaver, 2021-07-26:
Fix confirmed.  Thanks!

PR #1961 (2021-07-26): Choice button: Fix internal 'toggles' when labels change
Fixes #1960
commit aad0669e (2021-07-26): Choice button: Fix internal 'toggles' when labels change

Fixes #1960