ControlSystemStudio/phoebus #1965: Automated Testing of Displays

lorenzo-gomez-windhover, 2021-08-04:
I'm just curious. Is there a way of testing widgets/displays in an automated way?

Thanks
Lorenzo

kasemir, 2021-08-04:
See #36 and #314.

In principle, yes. Similar to unit tests for "normal" code, you can use testfx for UI tests.
You can create a display, simulate button pushes, verify if some aspect of the UI changed in response to the button push.
But there are some details of how to execute those tests on the various OS, and changes in pixel locations can mess with the test result, and overall so far nobody has had time to set that up and keep it running.
By all means give it a go.

lorenzo-gomez-windhover, 2021-08-04:
Thanks! [TestFX](https://github.com/TestFX/TestFX) looks interesting.

kasemir, 2021-08-04:
Well, I think when we last tried, it started out promising but there were several little issues.
To click a button, should you simulate a physical mouse move and click, or directly invoke the action event on a button?
When running the test, it would actually open a display, move the actual mouse pointer etc. That worked fine with a real display, but trying to run it with a virtual display within a CI build/test process was harder.
Still, we last tried that ~4 years ago, so by all means try it now!

jwlodek, 2026-10-19:
Hi all, if there's still interest I've started building an offline linter for phoebus screens: https://github.com/NSLS2/phoebuslint.

It catches issues like unexpanded macros, font sizes being too large for their widget parent, out of bounds widgets etc. It runs offline, by reading/parsing the underlying bob files, and supports auto-fixing a lot of the common problems. For now it's what I'd describe as an early prototype, but hoping to get it pushed further during the current codeathon.

shroffk, 2026-10-20:
The linter by @jwlodek is a simple and effective away to check OPI's without getting too bogged down in the complexity of runtime behaviours.
While we need testfx ui unit tests... the linter can help us catch a lot of very common issues.

@jwlodek can we move the repo to controlsystemstudio?
We also want to explore how to expand the checks... names colors and fonts, the use to widget classes, etc...

jwlodek, 2026-10-21:
To move the repo over I guess I'd need to be added to the org? I am OK with moving it as long as I can keep admin rights to the repo. Definitely interested in having others try it out and suggest additional checks.

shroffk, 2026-10-21:
This can be done