ControlSystemStudio/phoebus #1966: alarm-server alerts for a PV disabled during the alarm delay

mgotz, 2021-08-05:
Hi,

in the alarm-server, if I add a PV with a delay and an enabling filter the system shows alarms when I feel it shouldn't.
The situtation ist thus:
I have a PV1 in the alarm tree which has an alarm delay (e.g. 5 s) and an enabling filter (say PV2 != 0). PV2 is 1 and PV1 has no alarm. Then, the following happens:

1. t = 0s, PV1 goes into some alarm state.
2. t = 2s, (before the delay expires) PV2 is set to 0, now PV1 is greyed out in the alarm tree
3. t = 5s, (delay expires) an alarm is shown for PV1, which was greyed out before and should be disabled.

This causes alarms in our system for PVs that should be disabled.
I can then acknowledge the alarm and PV1 will show as green in the tree. However, it should show as grey and disabled. Additional alarms from PV1 are ignored by the system, as they should be, because the PV is disabled. So, GUI and alarm-server behavior remain out of sync.

Regards,
Malte

kasemir, 2021-08-05:
Yes, seems wrong. "Disabled" should still be honored with delayed alarms. I'll look at it.

kasemir, 2021-08-06:
Created a unit test that duplicates the issue and fixed it.
Also tried various combinations of enabling/disabling a PV with delay either, manually of via a filter expression.
They all behaved as one would expect, so please try again.

mgotz, 2021-08-09:
That was quick, thanks. It looks good to me now. I did some sandbox testing and everything behaves as expected. I will put it on our test system and see how it behave with the real PVs, but I don't really expect any surprises.

closed 2021-08-09 by kasemir