ControlSystemStudio/phoebus #1974: mysql 'storage_engine' variable unknown

thomasms, 2021-08-10:
I am using the archive engine with MySQL (5.7) and am getting an ```Unknown system variable 'storage_engine'``` error when trying to use the setup MySQL.dbd file (https://github.com/ControlSystemStudio/phoebus/blob/master/services/archive-engine/dbd/MySQL.dbd).

A quick online search suggests this could be due to the removal of the ```storage_engine``` variable from [this source](https://confluence.atlassian.com/confkb/confluence-fails-to-start-with-error-unknown-system-variable-storage_engine-using-mysql-5-7-x-789090576.html).

*MySQL 5.7.x uses InnoDB by default, and has removed the storage_engine system variable.*

Removing this line in the code, fixes the issue for me.

```sql
# Switch from default MyISAM (no checks but faster)
# to InnoDB ('real' database but slower)?
# THE BELOW LINE IS THE PROBLEM
SET storage_engine=INNODB;
# SET storage_engine=MyISAM;
```

kasemir, 2021-08-10:
Turning that section into a comment. The file is an "# Example MySQL Setup" after all, so need to read, understand, adjust for local needs in any case.

kasemir, 2021-08-31:
Closing since the original issue has been fixed.

closed 2021-08-31 by kasemir