ControlSystemStudio/phoebus #1986: Split Dock Behavior

lorenzo-gomez-windhover, 2021-08-20:
Hi there,

I discovered a particular behavior that happens when splitting tabs in phoebus and I'm wondering if this is expected? Consider the following scenario:

1. Split a tab into left/right or top/bottom.
2. Phoebus crates a second Pane with no tabs, as expected.
3. Close the tab to the left(assuming left/right split).

Now we have a window with no tabs and just an empty Pane. However when I try to close the window by clicking the X button, nothing happens. I understand why it happens and this code is really what causes that to happen on `org.phoebus.ui.docking.DockPane`:

https://github.com/ControlSystemStudio/phoebus/blob/dad6183377658491da1b695d53cae98248b28c55/core/ui/src/main/java/org/phoebus/ui/docking/DockPane.java#L579

What seems to happen is that once the dock is split with `parent.setCenter(null);`, it does not belong to a scene anymore.

So, given the scenario above, the Pane is in a "Scene-less" state. And I think that's why the X button does not work in that state. 

I think I may be able to fix it, but I'm just wondering if this is intended behavior?

Hope this is clear enough.

Thanks
Lorenzo

kasemir, 2021-08-20:
What you describe is about secondary windows, right?
I can only reproduce with secondary windows, not the original, main window that has the menu bar.

What should happen in secondary windows is that they close when their last pane is closed.
That's what Georg added in "CSSTUDIO-1126 Close empty windows", https://github.com/ControlSystemStudio/phoebus/commit/58418023e2f4266fd3f3f88411e56263fecc324a#diff-fe0aeed648c2aa215cfd0f6f4b45de35b0588496d59ca2cce2923b9d9d4e5c60

lorenzo-gomez-windhover, 2021-08-20:
Yes, you are absolutely right. It only happens with secondary windows. The main window does not appear to have this problem.

georgweiss, 2021-08-23:
The intended behavior is of course to close a window when all tabs in a secondary window have been closed. I admit I did not test a use case with split panes.
@lorenzo-gomez-windhover, let me know if you intend to fix this or if you want me to take a look.

lorenzo-gomez-windhover, 2021-08-23:
I'm attempting to fix it here:https://github.com/WindhoverLabs/phoebus/commit/f7e734d579fee06cf11d4cb28a2c8d5ae2ccd54e

And I'm almost there, I think. But it'd be awesome if you could take a look since you are far more familiar with the code base than I am.

georgweiss, 2021-08-23:
Sure, I'll take it for a spin.

georgweiss, 2021-09-15:
@lorenzo-gomez-windhover, can we close this?

lorenzo-gomez-windhover, 2021-09-15:
Yes.

closed 2021-09-15 by lorenzo-gomez-windhover