ControlSystemStudio/phoebus #1988: Combo widget seems to not work with pva

georgweiss, 2021-08-24:
A mbbo record associated with the Combo widget (items from PV = true) behaves differently when using pva (compared to ca). When selecting an item from the dropdown, the string value is used as the new value, which triggers an exception as it is parsed as a string, see attached stack trace. This does not happen with ca (I assume the JCA lib will map string -> int). For the sake of comparison, a TextEntry will handle this differently and map the string to an int, see TextEntryRepresentation#L240.

``
java.lang.Exception: Cannot parse number from 'Latch ERR'
	at org.epics.pva.data.PVANumber.parseString(PVANumber.java:46)
	at org.epics.pva.data.PVAInt.setValue(PVAInt.java:114)
	at org.epics.pva.client.PutRequest.encodeRequest(PutRequest.java:138)
	at org.epics.pva.common.TCPHandler.sender(TCPHandler.java:179)
	at java.base/java.util.concurrent.FutureTask.run(FutureTask.java:264)
	at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1128)
	at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:628)
	at java.base/java.lang.Thread.run(Thread.java:834)
``

kasemir, 2021-08-24:
Not sure what the correct behavior is.
Should the widget perform the mapping, or the CA/PVA library?
CA does handle type conversions, actually on the server (IOC) side, but that's in part for historical reasons. Way back the vxWorks IOCs were often more powerful than the OPI hosts. That soon changed, and the guideline then became: Yes, the IOC (CA server) can do the data type conversions, but a well-behaved client should instead do that and only pass the "native" data type.
For PVA we started with: Must write the native type. I still think that's the better choice, so in this case the Combo widget which got by with writing the string for CA should instead do the type conversion, just like the TextEntry already does.
 
Next question: How?
If the enumeration values are `{ "Apple", "Pear", "Banana" }`, it's somewhat obvious that writing the string "1" should be handled by writing the value `1` as an enumeration index, or maybe "Pear" as the associated label.
But the enumeration values might be `{ "1", "2", "4", "16", "32" }`, so is writing "4" meant to use index 4 or label "4" (index 2)? There might not be a perfect answer, but we can at least make it consistent by having the Combo widget call the same `FormatOptionHandler.parse` as the Text Entry already does.

georgweiss, 2021-08-24:
Agree that - for the sake of symmetry - the Combo widget should implement the same strategy as Text Entry.

What other widgets do we need to consider?

kasemir, 2021-08-24:
In principle, everything that can encounter an enum, but radio buttons, choice box, boolean widgets are already based on index, what matters here are widgets that might have a string value and need to convert to enum index.
Text Entry is already handled. I think Combo is the only one left, since it also allows entering text like the Text Entry.

georgweiss, 2021-08-24:
The use case triggering this actually uses indices 1,2,4, which makes the drop-down look as below. We should maybe also consider "fixing" this. Or is that a separate ticket?

I can take care of adding the mapping code to the Combo widget.

kasemir, 2021-08-24:
I think that's separate, and it's an issue on the IOC end.

There is no need to define the ENUM with 'gaps' just because the indices won't map to the desired raw values.
If you have states labeled `{ "No latching", "Latch ERR", "Latch ERR and NOK" }` which should correspond to raw values `{1, 2, 4 }`, check the manual for the EPICS mbbo record, which can handle that mapping:

```
record(mbbo, "Options")
{
  field(ZRST, "No latching")
  field(ZRVL, "1")
  field(ONST, "Latch ERR")
  field(ONVL, "2")
  field(TWST, "Latch ERR and NOK")
  field(TWVL, "4")
  field(NOBT, "4")
}
```

The PV "Options" will then be of type ENUM and have those three options, corresponding to enum indices 0, 1, 2.
But the PV "Options.RVAL" as shown in the text update at the bottom will have value 1, 2, 4:

[image]

kasemir, 2021-08-24:
.. but that brings up a new issue: Using CA, the combo, choice button and radio buttons will correctly show the current value. The combo and choice buttons also allow writing the PV. The radio button, however, only "works" for the first two indices. When selecting "Latch ERR and NOK" on the radio button, it writes VAL (index) 3?!

georgweiss, 2021-08-25:
@kasemir, thanks or investigating. I will fix the Combo issue as proposed by you as my users must use pva.

closed 2021-09-06 by georgweiss