ControlSystemStudio/phoebus #1989: Radio button issue with enum

kasemir, 2021-08-24:
Run this record:
```
record(mbbo, "Options")
{
  field(ZRST, "No latching")
  field(ZRVL, "1")
  field(ONST, "Latch ERR")
  field(ONVL, "2")
  field(TWST, "Latch ERR and NOK")
  field(TWVL, "4")
  field(NOBT, "4")
}
```

.. then try to display and control it via channel access, using a combo, choice buttons and radio buttons.
Widget at the bottom shows "Options.RVAL".

[image]

All works as expected except when selecting "Latch ERR and NOK" on the radio button, it writes VAL (index) 3?!

#1988

georgweiss, 2021-08-25:
Using this record the numeric values written are 0,1,2 in my test.

kasemir, 2021-08-25:
Did you check both VAL and RVAL? VAL is 0, 1, 2. RVAL should be 1, 2, 4.

georgweiss, 2021-08-25:
You're right, my bad.
So RVAL is 1,2,4 as expected, both using ca and pva to write.

kasemir, 2021-08-25:
Right, so as mentioned in #1988 this is partially an issue on the IOC end.
The PV for the GUI and maybe other higher-level logic would be like that mbbo "Options" with 3 states and value 0, 1, 2, no 'gaps' in the list of states. If the mbbo has device support, that would typically write the RVAL (1, 2, 4) out to hardware. If other database logic uses the data, the pulling record needs to read Options.RVAL, or if you want to push data from Options, you need to use DTYP="Raw Soft Channel".

georgweiss, 2021-08-25:
Then I am confused. If the radio button writes the index values and this results in the expected RVAL values, how is this an issue on the Phoebus radion button side? IOC implementation needs to consider whether to consume RVAL or not, and how to use the indices to be able to use the Combo box with no gaps. All in all, IOC engineer must be in sync with opi developer.

kasemir, 2021-08-25:
I've seen the radio button write the wrong index,  0, 1, 3(!) instead of 0, 1, 2.
The combo and the choice button always write 0, 1, 2.
All widgets update correctly.
The text update for RVAL shows the expected 1, 2, 4.
But the rightmost radio button sometime writes the wrong index 3 instead of 2.
Haven't figured out when exactly.

georgweiss, 2021-08-25:
OK, now I get it.

closed 2021-08-31 by kasemir