ControlSystemStudio/phoebus #1997: Cannot resolve org.epics:pbrawclient:0.0.8

nugatritter, 2021-08-30:
When I import the current phoebus project (585095) into Intellij IDEA, I get following error:
Cannot resolve org.epics:pbrawclient:0.0.8
How can I resolve this issue?

georgweiss, 2021-08-30:
Is this the only failing dependency? Can you build on command line, e.g. ``mvn clean install``? Any additional debug information available?
Screen shot from my IntelliJ instance showing resolved pbrawclient dependency:
[image]

nugatritter, 2021-08-30:
Yes, this is the only failing dependency.
I work under Windows and have not configured my command line to create Maven projects. Therefore, I cannot create these projects here.
Here are my dependencies in Intellij IDEA
[image]

nugatritter, 2021-08-30:
Okay, I set up Maven and here is the resulting log file:
```
D:\IdeaProjects\phoebus>mvn clean install
[INFO] Scanning for projects...
[WARNING]
[WARNING] Some problems were encountered while building the effective model for org.phoebus:phoebus-target:jar:4.6.6-SNAPSHOT
[WARNING] 'dependencies.dependency.(groupId:artifactId:type:classifier)' must be unique: com.atlassian.commonmark:commonmark:jar -> duplicate declaration of version 0.17.0 @ line 341, column 17
[WARNING] 'dependencies.dependency.(groupId:artifactId:type:classifier)' must be unique: com.atlassian.commonmark:commonmark-ext-gfm-tables:jar -> duplicate declaration of version 0.17.0 @ line 346, column 17
[WARNING] 'dependencies.dependency.(groupId:artifactId:type:classifier)' must be unique: com.atlassian.commonmark:commonmark-ext-image-attributes:jar -> duplicate declaration of version 0.17.0 @ line 351, column 17
[WARNING]
[WARNING] Some problems were encountered while building the effective model for org.phoebus:app-trends-rich-adapters:jar:4.6.6-SNAPSHOT
[WARNING] 'dependencies.dependency.(groupId:artifactId:type:classifier)' must be unique: org.phoebus:app-email-ui:jar -> duplicate declaration of version 4.6.6-SNAPSHOT @ line 25, column 17
[WARNING]
[WARNING] It is highly recommended to fix these problems because they threaten the stability of your build.
[WARNING]
[WARNING] For this reason, future Maven versions might no longer support building such malformed projects.
[WARNING]
[INFO] ------------------------------------------------------------------------
[INFO] Reactor Build Order:
[INFO]
[INFO] phoebus (parent)                                                   [pom]
[INFO] dependencies                                                       [pom]
[INFO] install-jars                                                       [jar]
[INFO] phoebus-target                                                     [jar]
[INFO] core                                                               [pom]
[INFO] core-vtype                                                         [jar]
[INFO] core-formula                                                       [jar]
[INFO] core-framework                                                     [jar]
[INFO] core-types                                                         [jar]
[INFO] core-util                                                          [jar]
[INFO] core-pva                                                           [jar]
[INFO] core-pv                                                            [jar]
[INFO] core-security                                                      [jar]
[INFO] core-ui                                                            [jar]
[INFO] core-logbook                                                       [jar]
[INFO] core-email                                                         [jar]
[INFO] core-launcher                                                      [jar]
[INFO] app                                                                [pom]
[INFO] app-diag                                                           [jar]
[INFO] app-log-configuration                                              [jar]
[INFO] app-probe                                                          [jar]
[INFO] app-pvtable                                                        [jar]
[INFO] app-pvtree                                                         [jar]
[INFO] app-update                                                         [jar]
[INFO] app-email                                                          [pom]
[INFO] app-email-ui                                                       [jar]
[INFO] app-errlog                                                         [jar]
[INFO] app-filebrowser                                                    [jar]
[INFO] app-logbook                                                        [pom]
[INFO] app-logbook-inmemory                                               [jar]
[INFO] app-logbook-ui                                                     [jar]
[INFO] app-logbook-olog                                                   [pom]
[INFO] app-logbook-olog-client                                            [jar]
[INFO] app-logbook-olog-client-es                                         [jar]
[INFO] app-logbook-olog-ui                                                [jar]
[INFO] app-logbook-elog                                                   [jar]
[INFO] app-rtplot                                                         [jar]
[INFO] app-databrowser                                                    [jar]
[INFO] app-display                                                        [pom]
[INFO] app-display-model                                                  [jar]
[INFO] app-display-representation                                         [jar]
[INFO] app-3d-viewer                                                      [jar]
[INFO] app-display-representation-javafx                                  [jar]
[INFO] app-display-fonts                                                  [jar]
[INFO] app-display-runtime                                                [jar]
[INFO] app-display-editor                                                 [jar]
[INFO] app-display-convert-medm                                           [jar]
[INFO] app-display-convert-edm                                            [jar]
[... cut ...]
[ERROR] -> [Help 1]
[ERROR]
[ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR]
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/MojoFailureException
[ERROR]
[ERROR] After correcting the problems, you can resume the build with the command
[ERROR]   mvn <args> -rf :core-framework
````

georgweiss, 2021-08-30:
Seems a junit test is failing the build.
Please run ``mvn -DskipTests=true clean install`` to establish that build is OK with respect to dependencies.

nugatritter, 2021-08-30:
That looks good.
```
D:\IdeaProjects\phoebus>mvn -DskipTests=true clean
[INFO] Scanning for projects...
[WARNING]
[WARNING] Some problems were encountered while building the effective model for org.phoebus:phoebus-target:jar:4.6.6-SNAPSHOT
[WARNING] 'dependencies.dependency.(groupId:artifactId:type:classifier)' must be unique: com.atlassian.commonmark:commonmark:jar -> duplicate declaration of version 0.17.0 @ line 341, column 17
[WARNING] 'dependencies.dependency.(groupId:artifactId:type:classifier)' must be unique: com.atlassian.commonmark:commonmark-ext-gfm-tables:jar -> duplicate declaration of version 0.17.0 @ line 346, column 17
[WARNING] 'dependencies.dependency.(groupId:artifactId:type:classifier)' must be unique: com.atlassian.commonmark:commonmark-ext-image-attributes:jar -> duplicate declaration of version 0.17.0 @ line 351, column 17
[WARNING]
[WARNING] Some problems were encountered while building the effective model for org.phoebus:app-trends-rich-adapters:jar:4.6.6-SNAPSHOT
[WARNING] 'dependencies.dependency.(groupId:artifactId:type:classifier)' must be unique: org.phoebus:app-email-ui:jar -> duplicate declaration of version 4.6.6-SNAPSHOT @ line 25, column 17
[WARNING]
[WARNING] It is highly recommended to fix these problems because they threaten the stability of your build.
[WARNING]
[WARNING] For this reason, future Maven versions might no longer support building such malformed projects.
[WARNING]
[INFO] ------------------------------------------------------------------------
[INFO] Reactor Build Order:
[INFO]
[INFO] phoebus (parent)                                                   [pom]
[INFO] dependencies                                                       [pom]
[INFO] install-jars                                                       [jar]
[INFO] phoebus-target                                                     [jar]
[INFO] core                                                               [pom]
[INFO] core-vtype                                                         [jar]
[INFO] core-formula                                                       [jar]
[INFO] core-framework                                                     [jar]
[INFO] core-types                                                         [jar]
[INFO] core-util                                                          [jar]
[INFO] core-pva                                                           [jar]
[INFO] core-pv                                                            [jar]
[INFO] core-security                                                      [jar]
[INFO] core-ui                                                            [jar]
[INFO] core-logbook                                                       [jar]
[INFO] core-email                                                         [jar]
[INFO] core-launcher                                                      [jar]
[INFO] app                                                                [pom]
[INFO] app-diag                                                           [jar]
[INFO] app-log-configuration                                              [jar]
[INFO] app-probe                                                          [jar]
[INFO] app-pvtable                                                        [jar]
[INFO] app-pvtree                                                         [jar]
[INFO] app-update                                                         [jar]
[INFO] app-email                                                          [pom]
[INFO] app-email-ui                                                       [jar]
[INFO] app-errlog                                                         [jar]
[INFO] app-filebrowser                                                    [jar]
[INFO] app-logbook                                                        [pom]
[INFO] app-logbook-inmemory                                               [jar]
[INFO] app-logbook-ui                                                     [jar]
[INFO] app-logbook-olog                                                   [pom]
[INFO] app-logbook-olog-client                                            [jar]
[INFO] app-logbook-olog-client-es                                         [jar]
[INFO] app-logbook-olog-ui                                                [jar]
[INFO] app-logbook-elog                                                   [jar]
[INFO] app-rtplot                                                         [jar]
[INFO] app-databrowser                                                    [jar]
[INFO] app-display                                                        [pom]
[INFO] app-display-model                                                  [jar]
[INFO] app-display-representation                                         [jar]
[INFO] app-3d-viewer                                                      [jar]
[INFO] app-display-representation-javafx                                  [jar]
[INFO] app-display-fonts                                                  [jar]
[INFO] app-display-runtime                                                [jar]
[INFO] app-display-editor                                                 [jar]
[INFO] app-display-convert-medm                                           [jar]
[INFO] app-display-convert-edm                                            [jar]
[INFO] app-display-navigation                                             [jar]
[... cut ...]
[INFO] service-alarm-logger 4.6.6-SNAPSHOT ................ SUCCESS [  0.032 s]
[INFO] service-alarm-config-logger 4.6.6-SNAPSHOT ......... SUCCESS [  0.015 s]
[INFO] service-save-and-restore 4.6.7-SNAPSHOT ............ SUCCESS [  0.016 s]
[INFO] ------------------------------------------------------------------------
[INFO] BUILD SUCCESS
[INFO] ------------------------------------------------------------------------
[INFO] Total time:  0.972 s
[INFO] Finished at: 2021-08-30T11:18:25+02:00
[INFO] ------------------------------------------------------------------------
```

nugatritter, 2021-08-30:
Okay, the missing dependency is gone. When I run phoebus out of Intellij I get the following message:
`Error: JavaFX runtime components are missing, and are required to run this application`
Do you have any advice on how to solve the problem?

georgweiss, 2021-08-30:
Added screen shot of my run configuration. Make sure "Use classpath of module" says "product". All other settings in the screen shot are up to you, except Main Class, which must be ``org.phoebus.product.Launcher``.

[image]

nugatritter, 2021-08-30:
`org.phoebus.product.Launcher`
That solved my problem. Everything works now. Thank you very much!

closed 2021-08-30 by nugatritter

georgweiss, 2021-08-30:
FYI: building with Maven installs pbrawclient to the local Maven repo.