ControlSystemStudio/phoebus #1998: action event for graphics

nugatritter, 2021-08-30:
I see the "Actions" property for the "Graphics" items and I can configure it. But when I run the bob file, I cannot trigger it.
```
<widget type="rectangle" version="2.0.0">
    <name>Rectangle</name>
    <x>270</x>
    <y>90</y>
    <width>110</width>
    <height>30</height>
    <actions>
      <action type="open_display">
        <file>open_valve.bob</file>
        <target>tab</target>
        <description>Open Display</description>
      </action>
    </actions>
  </widget>
```
Of course the file open_valve.bob exists. And when I use a button with this action, it works.

kasemir, 2021-08-30:
> I cannot trigger it

How did you try to trigger it? Did you open the context menu (right-click on the rectangle)?
See https://github.com/ControlSystemStudio/phoebus/wiki/Display-Builder-Compatibility#actions-action-buttons

nugatritter, 2021-08-30:
Thank's for the quick answer. Yes, right-click is working fine and I can open the target display.
I had expected an on-click event. Would this also be feasible? The background is that we want to use the panel later in the web browser with touch control.

kasemir, 2021-08-30:
Again, see https://github.com/ControlSystemStudio/phoebus/wiki/Display-Builder-Compatibility#actions-action-buttons

nugatritter, 2021-08-30:
Mea culpa.

closed 2021-08-30 by nugatritter

shroffk, 2021-08-30:
At NSLS2 we had ( are having ) a similar discussion since this functionality *was* available in BOY.

The argument against adding "on click" actions to widgets which don't explicitly give UI clues that an action will be performed is that it makes screens unpredictable and...can be dangerous.
Clicking on a icon of a valve can results in it being opened or closed... when the user might have just clicked on it to see the pv name or value.

We have decided to use context menu to force users to explicitly select an action before running it... it is 1 extra click but makes screen behaviour more reliable and predictable.

P.S. even though the documentation mentions transparent buttons, we are strongly discouraging the use of that too... context menus or adjacent buttons is the way we are hoping to go