ControlSystemStudio/phoebus #1999: Best Practices For Phoebus Products

lorenzo-gomez-windhover, 2021-08-30:
Hi there,

hope you are all doing well.

It's very likely that my team and I will be using Phoebus to make our own product in the future. And I was just wondering if there are any general tips/best practices you guys hope that people who make their own products follow? Something I should mention is that I think we might seriously consider having a fork of Phoebus rather than pull the dependencies from maven(which if I understand correctly,  are _not_ the latest and greatest from master ). We'll obviously keep our code decoupled from core and app modules. Essentially we'll have a maven sub-module just like the apps are organized in your repo at the moment. But it would also be nice to just have a fork in case we find bugs, we can always push back to your main to help project be better.

Hope these questions and concerns are clear.

Thanks in advance
Lorenzo

kasemir, 2021-08-30:
Fork of the Phoebus sources, maven dependencies, and own product seem separate aspects.

For sources, you should almost always be able to use the latest phoebus sources from master for all the common code.
You might mirror those to a local host as part of some build pipeline, but I'm not aware of any issues in the common code that have so far required any site to operate from a fork.

For dependencies like javafx, rxjava, jetty, ..., you have to get them somehow, see https://github.com/ControlSystemStudio/phoebus/tree/master/dependencies
We list the ones we use in https://github.com/ControlSystemStudio/phoebus/blob/master/dependencies/phoebus-target/pom.xml, so maven can fetch them. If you find one listed there that's out of date and we're missing out on some new feature or bug fix, make a specific suggestion as to which one to update. Everybody is of course free to fetch dependencies by other means, and in addition to building with maven we also support building with ant and various IDEs, but details differ by site, maven is the common denominator.

As for a site-specific product, yes, that does need its own repository where you put your build files like the ones shown in https://github.com/ControlSystemStudio/phoebus/tree/master/phoebus-product or https://github.com/shroffk/nsls2-phoebus or https://github.com/kasemir/phoebus-sns

shroffk, 2021-09-13:
> If you find one listed there that's out of date and we're missing out on some new feature or bug fix, make a specific suggestion as to which one to update.

+1
We want to try our best to stay on top of our dependencies.
Having the latest version of a library might be essential if there is security patch or an important missing feature... but each time we update a dependency we should be doing through testing to ensure everything continues to work. By creating issues tickets and we can leverage the effort of the collaboration and try to keep Phoebus as up to date as possible.