ControlSystemStudio/phoebus #2010: Question: default value for org.csstudio.trends.databrowser3/automatic_history_refresh

georgweiss, 2021-09-15:
Users suggest this setting should default to ``true`` as long-running data browser plots will draw horizontal lines when samples from the live buffer are being flushed.

Is there any reason this defaults to ``false``? The additional requests to the archiver if set to ``true`` should not cause unacceptable load on network and archiver...

kasemir, 2021-09-15:
It's more for historic reasons to have the original behavior by default.
Feel free to update the default, or set it in your site-specific product as I do for SNS,
https://github.com/kasemir/phoebus-sns/blob/737beec0ca03ab1e3a0d81d98e3f1a98be32eb73/product-sns/settings.ini#L79

shroffk, 2021-09-15:
While I have it set to true in my site specific preferences, I do feel like we might want to set it `true` by default.

The data retrieval and flushing of the buffer has been implemented in away that there are no bursts of archiver requests...given that is it quite common to have plots open for a long time I feel like changing the default to `true` would be better

georgweiss, 2021-09-15:
Thanks, I will create a PR.

closed 2021-09-15 by georgweiss