ControlSystemStudio/phoebus #2015: Cannot use ChannelFinder without credentials for read-only

juanfem, 2021-09-23:
I am trying to use the ChannelFinder application in Phoebus without defining any username and password, but I am getting a 401 error:
`com.sun.jersey.api.client.UniformInterfaceException: GET https://*********.esss.lu.se/resources/channels?~name=PV returned a response status of 401 Unauthorized`

Digging a bit into the code, it seems the issue is related to the flag `withHTTPAuthentication` in `org.phoebus.channelfinder.ChannelFinderClientImpl`, which is enabled by default and will try to get some default username and password from the properties file in the resources if those are not defined in the phoebus.ini file. Setting username and password properties to an empty string doesn't help either.

I couldn't find any setting to disable the `withHTTPAuthentication` flag, and I think it would be very convenient in this case.

kasemir, 2021-09-23:
I think the underlying motivation here is that there is no read-only client.
The client implementation can always not only read but it will also support write access, and for write access you naturally want to use some user/password.
So if you want to allow read access without any authentication, you'd have to introduce a read-only mode.

shroffk, 2021-09-23:
@kasemir is correct. This was the behaviour original intended and inherited from the old client. Moving forward I think I see some value in supporting a read only mode.

In your case can you try to use `http` instead of `https`, you might have to add the port too. In the meanwhile I will look into specking out how to support read only mode.

juanfem, 2021-09-23:
Ok, I think we had a mistake in our internal documentation on how to generate the ini file. I was assuming a read-only mode was available.

I tried with `http` and I get a 302 error instead, the server redirect to `https` I think. Maybe we can change that.

kasemir, 2022-05-25:
Read-only mode now addressed in #2070

closed 2022-05-25 by kasemir