ControlSystemStudio/phoebus #2016: Request: Confirmation dialogs for Text Entry widgets

keara-soloway, 2021-09-23:
Text Entry widgets do not have the option to add a confirmation dialog, which is available on many other widgets in the "Controls" category. Can this property be added?

kasemir, 2021-09-23:
In the last 30 years, edd/dm/edm/medm/BOY didn't need a confirmation dialog on text entry fields.
In general, friction from popup dialogs should be limited, since they prevent interacting with the display.
The text entry already has a little friction because one needs to press "Enter".
When writing is not possible because of access permissions, you get a "disabled" look and cursor that indicates it.
If you need to restrict access to text entries, you can place them on an "expert" sub-screen that's opened via a button with confirmation dialog.

keara-soloway, 2021-09-25:
Okay, thank you for the information!
My colleagues still agree that they want a confirmation dialog for some text entries, even with IOC access security, etc. in place. Since there are no plans to formally add them to this kind of widget, I'll implement these special cases with a script instead.
Thanks again for your reply.

closed 2021-09-25 by keara-soloway

kasemir, 2021-09-25:
For the related use case of editing a bunch of settings via text entry fields, requiring confirmation before writing the changes, with an automated logbook entry that documents who changed what from which original value to what final value, see PACE https://control-system-studio.readthedocs.io/en/latest/app/pace/doc/index.html