ControlSystemStudio/phoebus #2036: Multi Layer Display

htoraby, 2021-10-17:
Hi all, I would like to know of display could support multi-layer display which each widget could assign to the specific layer and control layer display and also zooming and zoom out based layer display. do you think about that?

kasemir, 2021-10-18:
Widgets are ordered back-to-front. There is no support for layers, but you can 'group' widgets and then show/hide a complete group. I would, however, generally advise against having widgets in the display and then not showing them, because  it means they're initially there, then require a script or rule to hide them, so it's overall more effective to simply not have them in the first place. Check out the "Navigation Tabs" for a more efficient way to change out part of a display.
There is the option to zoom in/out of a display.

htoraby, 2021-11-27:
Hi, I checked the idea of group widgets, can work but not feasible, it will be very complex screen, I think if the issue of multi layer added to the designer could be best option.