ControlSystemStudio/phoebus #2050: How to pass pv value to button "Execute Command"

tynanford, 2021-11-02:
I would like to pass command line arguments to a python program via a Phoebus action button. In particular, I'd like to pass local PV values from the screen as arguments. Is there a good way to do this with the execute command action? Attached is an example screen and python file. Thanks.

[button.zip](https://github.com/ControlSystemStudio/phoebus/files/7462212/button.zip)

kasemir, 2021-11-02:
Well, if you want to use a script to do unspeakable things to the GUI, but have to ask how, then maybe better not do that ;-) ?

Fundamentally you should be able to just pass a local PV into your script the usual way, then do something like this:

```
from ... import PVUtil
import os

value = PVUtil.getString(pvs[0])
os.system("python /path/to/that/script.py  " + value)
```

But starting an external program could result in resource leaks, so watch out for that. Check if the overall process keeps growing in size each time you trigger the script

tynanford, 2021-11-02:
I blame @shroffk ! ;)

Thanks, I will use the script solution. I'm looking into building a screen in python with a button press in Phoebus.

closed 2021-11-02 by tynanford

kasemir, 2021-11-02:
Kunal is not only perfect in general, we also often find ourselves in agreement, so this is inconceivable. You must be talking about a different person.

tynanford, 2021-11-02:
It must have been a brief moment of temporary insanity :)