ControlSystemStudio/phoebus #2062: add possibility to display legend for only selected PVs in a plt

olivierdelahaye, 2021-11-18:
When displaying a PLT with several PVs, you have the possibility to show or not the graph of a PV with the Properties Table.
When you display the legend, all the PVs are displayed in the legend even not selected in the properties table.
Is it possible to display only the legend for the selected PVs ?
In a same way, on the right of the graph, all the PV names are displayed even not selected in the properties panel. It would be nice to see only selected PVs on the properties panel.
See below.

[image]
t

kasemir, 2021-11-18:
The fact that 'hidden' traces still show up in the legend or on the right axis is a bug. Do you have time to try and fix it?

olivierdelahaye, 2021-11-18:
I can have a look on it. I keep you updated.

tynanford, 2021-12-21:
We had a few users notice this issue recently as well. I went ahead and created a pull request for a possible fix here: https://github.com/ControlSystemStudio/phoebus/pull/2099

kasemir, 2022-01-03:
Tynan just fixed it, thanks

closed 2022-01-03 by kasemir