ControlSystemStudio/phoebus #2063: Web browser widget slow loading page

charlesHenryPATARD, 2021-11-19:
Hello,

We're using CS-Studio Phoebus for Spiral2 command control.

I'm developping a BOB app which use the Web Browser Widget in a Tab Container.

In our developpement process we have two steps : 

1. Development : Start app with Start.bob
2. Deployment : Apps are started in embedded display widget which call the development Start.bob in step 1

 Loading  web page in step 1 : everything is ok.
 Loading web page through embedded display widget is wery slow : 20 secondes

I put files in a zip archive. 

- Start.bob is file to start in setp 1
- BeamDump.bob is the Embedded diplay bob which call Start.bob 
- userdoc.html is the file loadded in web web browser
- Web Browser is located in Tab named "Aide"

[SlowWebBrowser.zip](https://github.com/ControlSystemStudio/phoebus/files/7569559/SlowWebBrowser.zip)

kasemir, 2021-11-19:
The example doesn't work for me. It expects the html file in /home/phoebus/IHM/Reglages/BeamDump/Ressources/readme/userdoc.html, and the embedded display in ../../../IHM/Reglages/BeamDump/Start_beamdump.bob
In addition, the web page contains a lot of javascript.
Bottom line might be that this is just too much for the web widget and you're better off viewing the web page in a normal web browser. For example, use an action button with "Open Web Page" action to open in a normal web browser.

charlesHenryPATARD, 2021-11-19:
Hi,

Sorry, I forget to change paths in example files. Now it's ok except for userdoc.html. You have to mentionned the full absolute path like :    file:///absolute_path/to/userdoc.html

The html file is a Markdown to html convertion.

As I just said before when you start with Start.bob it's really quick. When you start with BeamDump.bob it's really slow and no execptions are raised.
I also tested to load file with firefox and it's works fine.

I attach the new version of example

[SlowWebBrowser.zip](https://github.com/ControlSystemStudio/phoebus/files/7570567/SlowWebBrowser.zip)
.

kasemir, 2021-11-19:
The updated example, when I edit Start.bob to include the full path tot he userdoc.html, works fine for me.
I see no noticeable difference in opening Start.bob vs. BeamDump.bob.
This is on Mac OS X.

Running under jProfiler, when I keep opening and closing Start.bob, then BeamDump.bob, that again Start.bob and so on, most of the CPU usage is in the WebView init. So simply creating the web view is expensive. I'm not sure what's really behind that. With the SWT implementation, the web widget called out to a `xulrunner` library which loads some native web runtime. We had trouble with that, where an update of Firefox would break the web displays inside SWT. On the other hand, with web browsers evolving so rapidly, it makes sense to call out to one of those implementations instead of having SWT or JavaFX try and implement their own web renderer. So bottom line, the SWT web widget has been problematic and the JavaFX web widget might also have issues. 
Still, I cannot reproduce the slow loading of the 'embedded' example in  BeamDump.bob. Certainly not 20 seconds, in fact I really don't see a difference at all.

[image]

charlesHenryPATARD, 2021-11-22:
Hello,
Thank you Kay for your answer.

We're using Linux CentOS 7. It's maybe a graphical issue. We're missing time and man power to investigate further. 
I'll try to find another solution as you said (Action -> Open in system web browser).

georgweiss, 2021-12-17:
Have you been able to resolve this? If so, can we close this?

charlesHenryPATARD, 2021-12-17:
Hi,
Resolve this ? Not really. Due to a lack of time.
To avoid this I open page in external Web browser.

You can close issue.

georgweiss, 2021-12-17:
OK, thanks.

closed 2021-12-17 by georgweiss